MarlinFirmware/Marlin #21720: [BUG] Marlin 2.0.x Does not compile with SLIM_LCD_MENUS and DISTINCT_E_FACTORS enabled

lujios, 2021-04-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello.
When trying to free up space from PROGMEM, I turned on SLIM_LCD_MENUS. But the compilation fails with the following error.
  
[image]

I found that '_planner_refresh_e_positioning' is defined in the same file associated with the error, "menu_advanced.cpp", and that it was linked to the active DISTINCT_E_FACTORS. But it also depended on SLIM_LCD_MENUS being disabled. That is, as the original code is, either SLIM_LCD_MENUS is disabled, or it is not possible to define the function '_planner_refresh_e_positioning'.
   
[image]

As I needed both configurations, I added a '#else' to the '#if' associated with the SLIM_LCD_MENUS validation disabled to always define '_planner_refresh_e_positioning', like this;
   
[image]

As a result, the compilation finishes OK, with both settings active.
There is mention of this same error in #18147, but no solution was provided.
I can assume there will be a more refined solution, but this seemed the fastest to me.

### Bug Timeline

There is mention of this same error in #18147, but no solution was provided.

### Expected behavior

Enable SLIM_LCD_MENUS and DISTINCT_E_FACTORS, and compile without failures.

### Actual behavior

Compilation error.

### Steps to Reproduce

1.  Enabled DISTINCT_E_FACTORS.
2.  Enabled SLIM_LCD_MENUS.
3.  Compile.

### Version of Marlin Firmware

Marlin-bugfix-2.0.8

### Printer model

Delta Anycubic linear.

### Electronics

Trigorilla V1.0 Mega2560.

### Add-ons

Drivers TMC2209. 

### Your Slicer

Other (explain below)

### Host Software

Repetier Host

thisiskeithb, 2021-04-27:
Since you've provided a fix, it'd be faster if you submitted a [Pull Request](https://marlinfw.org/docs/development/getting_started_pull_requests.html) so it can be reviewed.

lujios, 2021-04-27:
Just a complete newbie at this. Can you tell me how to submit a Pull Request?
Thanks.

thisiskeithb, 2021-04-27:
> Just a complete newbie at this. Can you tell me how to submit a Pull Request?
> Thanks.

There's a link in my last reply.

ellensp, 2021-04-30:
PR has been merged

closed 2021-04-30 by ellensp