MarlinFirmware/Marlin #21721: Change font/bootscreen/statusScreen on marlin 2.0.x

00markin, 2021-04-27:
### Is your feature request related to a problem? Please describe.

I've tried to follow [Marlin's doc](https://marlinfw.org/docs/development/fonts.html) to change the font of menu in Marlin 2.0.x and couldn't change it... is there any other way to do it? Also trying to change the bootscreen without success... Also want to customize statusScreen with gcode preview... Can someone help-me?

### Describe the feature you want

Way to add custom fonts, bootscreen and statusScreen

thisiskeithb, 2021-04-27:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-04-27 by thisiskeithb

00markin, 2021-04-27:
Actually i'm trying to make marlin more customizable (developing tools for easy way to make marlin more functional for TT Bluer), and actually I've checked Marlin's doc (as described before...)

thinkyhead, 2021-05-02:
https://discord.gg/n5NJ59y