MarlinFirmware/Marlin #21722: [BUG] Nozzle moves too close to the bed with LCD bed leveling (PROBE_OFFSET_WIZARD) and separate Z_MIN_PROBE and Z_MIN_ENDSTOP

mildred, 2021-04-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin bugfix-2.0.x at commit 516fe131bea242be903d0ec6d90a8f6f209ef173

I configured a Creality Ender-3 firmware to use BLTouch keeping the z axis stop sensor separate from the bltouch sensor.

To calibrate the bltouch z offset, I go to the advanced settings menu, sensor offsets, and choose the wizard. When running, after homing on the front-left corner, the nozzle moves very close to the bed to the middle of the bed to run the wizard. It is so closes that it can scratch the bed. 

The configuration:

```c
    // #define USE_PROBE_FOR_Z_HOMING // disabled
    // #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN // disabled
    #define BLTOUCH
    #define NOZZLE_TO_PROBE_OFFSET { -44, -6, -3.37 } // use own values X,Y,Z
    #define AUTO_BED_LEVELING_BILINEAR
    #define HOMING_FEEDRATE_MM_M { (120*60), (120*60), (8*60) }
    #define NOZZLE_PARK_FEATURE
    #define LCD_BED_LEVELING
    #define SOFT_ENDSTOPS_MENU_ITEM // helps with calibrating z offset

    #define BABYSTEPPING 
    #define BLTOUCH_SET_5V_MODE
    #define ADVANCED_PAUSE_FEATURE
    
    #define PROBE_OFFSET_WIZARD
```

### Bug Timeline

first time I use this feature

### Expected behavior

- run wizard
- move to the front-left corner until the z axis sensor is triggered
- **move up a little bit (10mm?)**
- move to the center of the bed
- perform bltouch calibration

### Actual behavior

- run wizard
- move to the front-left corner until the z axis sensor is triggered
- **nozzle is not moved up, risking scratching the bed**
- move to the center of the bed
- perform bltouch calibration

### Steps to Reproduce

1. Flash firmware
2. Run z offset wizard

### Version of Marlin Firmware

516fe131bea242be903d0ec6d90a8f6f209ef173

### Printer model

Creality Ender 3

### Electronics

Board v4.2.2

### Add-ons

runout sensor, bltouch (not wired on z sensor which is kept)

### Additional information & file uploads

- [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6387003/Configuration.zip)
-

mildred, 2021-04-27:
Looking around, it seems to be in:

https://github.com/MarlinFirmware/Marlin/blob/e9618f1ba25c0560f490ce831e10d1906fa686b1/Marlin/src/lcd/menu/menu_probe_offset.cpp#L127

There is a G29 to home the nozzle but noting to move Z a little higher... I think we could add:

```
line_to_z(10);
```

strayr, 2021-05-02:
I think this might be the same bug as https://github.com/MarlinFirmware/Marlin/issues/21727 from a different perspective?

I'm just rewiring to try my probe as an endstop becasue USE_PROBE_FOR_Z_HOMING without Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is not applying the z offset to the home position so my Z is being stretched by the Z-offset across the Z fade height.

closed 2021-07-16 by github-actions[bot]