MarlinFirmware/Marlin #21724: Config Reversing Stepper Direction when adjusting feedrate and mixing stepper

jameswis01, 2021-04-28:
### Bug Description

TMC2209  - BLTouch - sensorless homing.  Works perfect.  Lower feedrates and acceleration .  Works Perfect.  Enable Mixing Steppers.  X AXIS REVERSES!!!!!   Comment Out Mixing Steppers, Recompile and issue still exists...  Tried with the most recent and last 4 versions of marlin... New configs everytime.  Can duplicate EVERYTIME!!!!
### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

Setup - X, Y, Dual Z, Dual Extruders, Mixing Nozzle
TMC2209's Sensorless, No stealth, no hyper. w/ BLTouch.
Configure settings. test.  Lower Feedrates & Acceleration.  X Axis Reverses homing direction.. After enabling Mixing Steppers.. Direction on touchscreen & LCD are still moving correctly.
  Disable mixing steppers, and issue persists.

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**
X axis homes in wrong direction everytime.  Moves in correct direction from LCD and Touchscreen.

**Actual behavior:**
read above.
[Buggy Marlin.zip](https://github.com/MarlinFirmware/Configurations/files/6373431/Buggy.Marlin.zip)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

thinkyhead, 2021-04-28:
So, for a mixing setup you will want to make sure your extra E stepper pins are not conflicting in weird ways. So, first download the latest `bugfix-2.0.x` and enable the `PINS_DEBUGGING` option and flash the firmware. Send `M43` and examine the output carefully to make sure everything looks correct. If you're not sure, then paste the output into a reply here.

thisiskeithb, 2021-11-25:
Closing since requested information was never provided and issue appears abandoned.

closed 2021-11-25 by thisiskeithb