MarlinFirmware/Marlin #21725: Tramming wizard: no clearance before moving between corners

HLammers, 2021-04-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I’m using the tramming wizard (G35), the nozzle is not raised to get clearance from the bed when moving between corners. This means that the probe is scratching the print bed surface.

A related issue is that if I set `Z_AFTER_PROBING` to 1, so the nozzle moves close enough to the bed for the probe to retract, the probe isn’t available for for the tramming wizard anymore. For the HallON v3 probe I’m using (https://shop.garage-makezone.eu/en_US/p/HallON-v3-Ender-3-Ender-3-Pro/43), the X axis need to home to deploy the probe and there doesn’t seem to be an option to make Marlin do so (I’m considering filing a feature request). So if I set `Z_AFTER_PROBING` to 1 I can’t use the tramming wizard).

### Bug Timeline

This is the first time I’m trying the tramming wizard

### Expected behavior

I expect the nozzle to raise before moving, for example using `Z_CLEARANCE_BETWEEN_PROBES` from the ABL settings.

### Actual behavior

The nozzle is moved without raising, scratching the probe over the print bed surface.

### Steps to Reproduce

1. Enable tramming wizard in Marlin (`#define ASSISTED_TRAMMING`)
2. Start the tramming wizard using the ‘L corner’ option in the ABL menu (on a BBT TFT35 E3 v3 touch screen)
3. Press start

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Apr 28 2021 12:09:23)

### Printer model

Ender 3 Pro

### Electronics

BBT SKR Mini E3 v1.2

### Add-ons

HallON v3 bed level sensor, BBT TFT35 E3 v3 touch screen (with firmware version TFT_E3_V3.0.27 Apr 27 2021)

### Your Slicer

Cura

### Host Software

OctoPrint

HLammers, 2021-05-01:
I found a solution by defining the probe differently, even though it is not (yet) supported by Marlin (see https://github.com/MarlinFirmware/Marlin/issues/21726). Now it works. The other issue is related to the TFT35 firmware, which doesn’t support the G35 command (or not correctly).

closed 2021-05-01 by HLammers

thinkyhead, 2021-05-02:
Apparently you can set `Z_CLEARANCE_BETWEEN_PROBES` to some value and it will be used in `G35`, but I don't know if it will be applied in exactly the right way for your setup.