MarlinFirmware/Marlin #21726: Deploy bed leveling probe by homing x axis

HLammers, 2021-04-28:
### Is your feature request related to a problem? Please describe.

For the HallON v3 probe I’m using (https://shop.garage-makezone.eu/en_US/p/HallON-v3-Ender-3-Ender-3-Pro/43), the x axis need to home to deploy the probe and there doesn’t seem to be an option to make Marlin do so.

### Are you looking for hardware support?

HallON v3 auto bed level probe by garage-makezone.eu (https://shop.garage-makezone.eu/en_US/p/HallON-v3-Ender-3-Ender-3-Pro/43)

### Describe the feature you want

An option to set deployment of the bed level probe by homing the x axis (after raising the z axis to `Z_HOMING_HEIGHT`), which is similar to how deployment of the TouchMe probe works, except that it doesn’t require to lower the z axis by 0.2 mm

### Additional context

This is related to this issue: https://github.com/MarlinFirmware/Marlin/issues/21725

thisiskeithb, 2021-04-28:
`TOUCH_MI_PROBE` may work:

https://github.com/MarlinFirmware/Marlin/blob/2e0a1f1afff71a29e5fb5eb99c509d0de5384087/Marlin/Configuration.h#L961-L976

HLammers, 2021-04-28:
Thank you @thisiskeithb, for your quick response.

I just tested enabling `TOUCH_MI_PROBE`, but I’m running into at least two issues when I try to compile:

The first one (which might not turn out to be a problem, if I would get it working) is that my trick to retract the probe doesn’t work:

> Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1359:8: error: #error "TOUCH_MI_PROBE requires Z_AFTER_PROBING to be disabled."

More problematic is this one:

> Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1363:8: error: #error "TOUCH_MI_PROBE requires Z_MIN_PROBE_ENDSTOP_INVERTING to be set to false."
 
The HallON v3 sensor requires `Z_MIN_PROBE_ENDSTOP_INVERTING` to be set to true

thisiskeithb, 2021-04-28:
Bummer. It shouldn't be too difficult to adapt for this probe.

ellensp, 2021-04-28:
The second one is just a sanity check for that TOUCH_MI_PROBE, you can comment it out to test it.

HLammers, 2021-04-28:
@ellensp, in which file?

ellensp, 2021-04-28:
Marlin/src/inc/SanityCheck.h line 1363

comment out these two lines.
```
    #elif Z_MIN_PROBE_ENDSTOP_INVERTING
      #error "TOUCH_MI_PROBE requires Z_MIN_PROBE_ENDSTOP_INVERTING to be set to false."
```

HLammers, 2021-04-28:
My first test seems to be positive: commenting out those lines in SanityCheck.h allows me to compile Marlin and it seems to work as it should on my printer. I’ve tested ABL and homing, which both works, including moving toward the bed to retract the probe. I’ll do some more testing (probably tomorrow).

HLammers, 2021-05-01:
I can confirm that it works, so for adding native HallON v3 support the Touch-Mi code can be copied, with only one difference in the SanityCheck.h file.

Minimum required settings for the HallON v3 are:

- `#define Z_MIN_ENDSTOP_INVERTING true`
- `#define Z_MIN_PROBE_ENDSTOP_INVERTING true`
- `#define USE_PROBE_FOR_Z_HOMING`
- `NOZZLE_TO_PROBE_OFFSET` depending on the set-up
- `#define Z_HOMING_HEIGHT   45`
- `#define X_MIN_POS -5` (reflecting the x end stop position, so the exact number might be different, depending on the printer model - if not set correctly, the probe might not be deployed)
- `#define Z_SAFE_HOMING`