MarlinFirmware/Marlin #21727: [BUG] Z probe offset ignored unless UBL is enabled

nelsongraca, 2021-04-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Running latest code from bugfix-2.0.x branch, can't test latest release since there is a fix to avoid watchdog restarting when saving UBL mesh.

If I set probe offset M851 Z0 and run G28 and then G0 Z0 the distance of nozzle to bed is around 2mm, if I set the Z offset M851 Z-1.5 then G28 and G0 Z0 the nozzle to bed distance is still around 2mm when should be 0.5mm.

If I then run G29 P0, G29 P2, G29 P3 (until the mesh is complete) G29 S0, G29 A. When I home with G28 and then G0 Z0 the distance of the nozzle to bed is 5mm.

G29 T shows a grid full of values around -1.5, I assume the grid is having the offset subtracted.

Current config attached.

More info or tests ask.

### Version of Marlin Firmware

hotfix-2.0.x

### Printer model

Ender 5 Pro

### Electronics

Stock Creality 4.2.2

### Add-ons

BL Touch connected to the 5 pin connector

nelsongraca, 2021-04-29:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/6400420/config.zip)

nelsongraca, 2021-04-30:
Something I just tested enabled Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and disabled USE_PROBE_FOR_Z_HOMING and changed the black and white wires from the dedicated bl touch connector to the Z switch, now everything works as expected.
The reported problem no longer happens, I wonder if it's some different behavior due to this config.

strayr, 2021-05-02:
I ran into this as well. A G29 J will get it to print correctly, but it makes calibration prints that don't need UBL a but annoying. I could rewire my printer to solve this, but it appears to be a regression from 2.0.7.2 behaviour

strayr, 2021-05-02:
A followup, G29J gets me a first layer on the bed, but my prints are stretched upwards by the Z-offset over the UBL fade distance. That's usually about 2mm for most bl-touch style setups over a default of 10mm which not only makes prints way off size wise but also look awful. I didn't notice the height was wrong until I compared it against the same print done with 2.0.7.2 era bugfix. 

This change is **recent** and affects everyone who uses a probe on dedicated pins, so the default install for ender 3 users.

[image]

A better summary is "Z offset is applied to mesh not home position when USE_PROBE_FOR_Z_HOMING is enabled, and it breaks everything"

strayr, 2021-05-02:
This also looks like the same issue https://github.com/MarlinFirmware/Marlin/issues/21584

thinkyhead, 2021-05-02:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

strayr, 2021-05-02:
Thanks for that @thinkyhead 
I've just found another and was working on documenting it, i'll redo with detailed logging. Going to have to make a quick conversion harness so i'm not rewiring connectors when chaning how the probe is wired, but something has gone really sideways.

spikedsoft, 2021-05-03:
I have the same problem when I use bilinear leveling.

I do the Leveling with G29, then Home with G28, Move to Z0, disable the Soft-Endstops, move down to the correct Z height and set that height as the Z-Offset with M851, but then after another G28 and Move to Z0 it is at the position where it was before, ignoring the offset. Even if i set something extreme like M851 Z10 it doesn't make any difference.

When using UBL it works (after G29 J, withouth G29 J it's the same problem)... But i guess with the same caveat reported by strayr.

Attached is my config as well as the output from the procedure you requested.

[Marlin_Config_Debug.zip](https://github.com/MarlinFirmware/Marlin/files/6416066/Marlin_Config_Debug.zip)

Let me know if you need more information. Also feel free to contact me on your Discord if you want, i'm loged in as "spikedsoft" on your Server.

strayr, 2021-05-03:
Ok, ive spent a lot of time on this trying to replicate, bugfix as of 3am today and 2.0.8 release work independent of how i configure my probe. 
I think i managed to replicate it **by not doing an M502 M500** after flashing firmware at some point but couldn't reproduce reliably.
I need to get in the habit of dumping "good" settings by M503 to a text file, and just running that as gcode on fresh firmware. And getting better at git so i can go back to an old firmware and config and know what it was built off.

spikedsoft, 2021-05-05:
Here is an update after receiving Discord Support:

The problem in my case has been that I used the regular Z-endstops instead of the probe for Z-homing. After changing it to use the probe it works as expected now.

The reason i was using the endstops is that the printer uses dual Z motors and dual Z endstops and i liked the automatic synchronization i got by using the endstops. I now enabled Z_STEPPER_AUTO_ALIGN instead and that works as well.

However, it makes things a bit slow, having to do G34 every time before the print. 

Is it intended that the probe Z-offset is only applied when the probe is also used for homing?

no-trick-pony, 2021-06-29:
> Here is an update after receiving Discord Support:
>
> The problem in my case has been that I used the regular Z-endstops instead of the probe for Z-homing. After changing it to use the probe it works [...]

Do you by any chance know if this is a new issue (or a version where this still worked for you)? On my machine with Marlin 2.0.6 (which is now ~1 year old) z-probe offset is ignored as well with BL touch, ABL_BILINEAR and dual z-motors. I can literally set any value for z-probe offset, save it, do the leveling, save that, print, and it will start printing 0.5mm to high, as always. I am using the z-endstops as well (mostly for synchronization and because I trust them more to not fail and cause a fatal crash). Right now I am looking through open and closed issues and this is the closest one I found so far. Since this issue here apparently concerns my exact setup, updating to the newest version of Marlin seems pointless as well. Not sure what to do.

spikedsoft, 2021-06-29:
> Do you by any chance know if this is a new issue (or a version where this still worked for you)? On my machine with Marlin 2.0.6 (which is now ~1 year old) z-probe offset is ignored as well with BL [...]

I don't know if it's a new issue.. The original Chiron firmware didn't have the problem but i don't know how that was configured... 

However, I also got a bit annoyed at having to run G34 before every print.... I found a solution that works for me.. Instead of tuning the Z-probe-offset, i now tune the Z-home position. I figured that out after quite a bit of trial and error... I never read about this method anywhere before so maybe it's wrong.. Use it at your own risk... :) But so far i didn't have issues and I can finally use the Z-endstops again in combination with BL-Touch leveling...

In order for this to work, you should make sure that your Z-home position is at 0 at the beginning (check it with M503).

Here is the procedure i do for leveling:

1. Heat up the bed, and manually level it via the 4 leveling screws...
2. Home and move to Z0 - G28 / G90 / G1 Z0
3. Now use Babystepping to move the nozzle to the correct zero position. Remember the displayed Z offset.
4. Update the Z-home by negating the value from step 4 - M206 Z<Value from step 3 * -1>
5. Run ABL

After step 5, you can test if it homes to the correct height.. But keep your finger on the power switch in case it crashes the bed... :) If it doesn't go to the position you previously set and you have soft Z endstops enabled, you may have to disable them...

After running that procedure, it should be approximately correct.. If it's not perfect, you can fine-tune the Z-home position by first resetting it to 0 (M206 Z0) and then doing a first-layer test print. While it's printing, babystep the Z-position until you're satisfied and again note the displayed Z offset. Then set the Z-home position again like in step 4...

DerAndere1, 2021-08-30:
I think this is not a bug and can be closed. 

> The problem in my case has been that I used the regular Z-endstops instead of the probe for Z-homing. After changing it to use the probe it works as expected now.

and

> Instead of tuning the Z-probe-offset, i now tune the Z-home position. [...] and I can finally use the Z-endstops again in combination with BL-Touch leveling...

Reading the fine manual, this all looks like expected behaviour and seems to be in accordance with the documentation at https://marlinfw.org/docs/hardware/endstops.html#configuring-endstops-and-probes.

> Is it intended that the probe Z-offset is only applied when the probe is also used for homing?

Yes, see documentation above.

closed 2021-08-30 by thisiskeithb