MarlinFirmware/Marlin #21728: E-Stepper not turning off not confined to Marlin 2.x...

Chromatamata, 2021-04-28:
Problem solved. A rather foolish typo caused this, sorry!

_Originally posted by @thinkyhead in https://github.com/MarlinFirmware/Marlin/issues/16753#issuecomment-581799822_

Hi,
I read a recent bug report which contained the above comment, concerning an issue where the e-stepper was not turning off in Marlin 2.x.

I have a printer with only 128k of memory and so use Marlin 1.9.

I recently came to hand code a filament change in some gcode I was writing and discovered the exact same issue in this version of Marlin which is why I started to browse for known bugs and solutions and stumbled across this post. You can send either m84 E or m18 E to the machine and neither stops the e-stepper motor's ironclad grip on the filament.

I see from the original post that it was due to a typo and has been fixed. I was wondering if you might be able to share what Marlin file that typo was in and what that typo was.

You see, I cannot use Marlin 2.x on my machine and retain all the functionality I want and I also don't want to try to use the 1.9 bugfix branch as I have quite a heavily modified code for my firmware and although I have annotated all my changes and personalisations, it would take me hours to track and find all those changes in order to recompile a new version. It would be very easy, however, for me to correct the appropriate part of the code, myself if I knew what it was and recompile.

thisiskeithb, 2021-04-28:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-04-28 by thisiskeithb

Chromatamata, 2021-04-28:
Uhmmmm? I'm sorry but I'M REALLY CROSS! I KNOW this is a bug and it's been happening on other versions of Marlin too as I linked to at the top of the submission. I am NOT asking for beginner advice on setting anything up!!!!!! I have reprogrammed part of my firmware MYSELF and don't need beginners help!!!!! Why on earth would you close this perfectly legitimate bug report? I am submitting this bug so that the developers can fix it and asking that they might kindly tell me the location of this problem in the code so I might fix it in my own firmware in the easiest possible manner for me... I don't think this is the wrong place for this.

thisiskeithb, 2021-04-28:
> and asking that they might kindly tell me the location of this problem in the code so I might fix it in my own firmware in the easiest possible manner for me... I don't think this is the wrong place for this.

It is. Please use the resources linked above.

descipher, 2021-04-28:
"I am NOT asking for beginner advice" ? 
Then apply it yourself using the info in the link you posted.
[image]