MarlinFirmware/Marlin #21732: [BUG] (CR10_STOCKDISPLAY not working on BTT SKR 2)

Blazin64, 2021-04-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When using `CR10_STOCKDISPLAY` on the latest `bugfix-2.0.x` branch, the display stays blank when the board is turned on. The copy in [BigTreeTech's repo](https://github.com/bigtreetech/SKR-2/) works just fine, however.

### Bug Timeline

New bug

### Expected behavior

After turning the board on, I expect to see the Marlin boot screen and then the menu.

### Actual behavior

The display stayed blank after the board turned on.

### Steps to Reproduce

1. Make the changes mentioned below to the default configuration.h and platformio.ini
2. Compile the firmware and put it on the SKR 2.
3. Turn on the board.

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Creality Ender 3 Pro

### Electronics

BTT SKR 2

### Add-ons

Stock display

### Host Software

SD Card (headless)

### Additional information & file uploads

I made the following changes in Marlin, matching BigTreeTech's serial and motherboard config:

configuration.h
```
#define SERIAL_PORT 1
#define SERIAL_PORT_2 -1
#define MOTHERBOARD BOARD_BTT_SKR_V2_0
#define CR10_STOCKDISPLAY
```

platformio.ini
```
default_envs = BIGTREE_SKR_2
```

I made the following changes in BigTreeTech's copy:

configuration.h
```
#define CR10_STOCKDISPLAY
```

Platformio.ini was already pre-configured in BigTreeTech's copy.

thisiskeithb, 2021-04-28:
I finally got a hold of a retail SKR 2 so I will take a look at this.

Blazin64, 2021-04-28:
It turns out the issue had nothing to do with the display. Even with displays disabled, USB serial does not work. This makes me think Marlin was not functioning at all.

The problem was actually caused by an outdated ststm32 version in one of Marlin's ini files. The ststm32 version under `[common_stm32]` in `ini/stm32f1.ini` is set to 12.0. That particular version has a RAM issue with the STM32F407VG, which is used on the SKR 2!

The RAM issue was fixed in version 12.1.0, according to the [stsm32 releases page](https://github.com/platformio/platform-ststm32/releases). I changed the version in `ini/stm32f1.ini` from 12.0 to 12.1. VS Code automatically downloaded the new version during compilation. I flashed the new firmware, and it appears to work!

thisiskeithb, 2021-04-28:
> The problem was actually caused by an outdated ststm32 version in one of Marlin's ini files. The ststm32 version under [common_stm32] in ini/stm32f1.ini is set to 12.0. That particular version has a RAM issue with the STM32F407VG, which is used on the SKR 2!

I just got my SKR 2 setup on the bench and can confirm that forcing ststm32 to 12.1 works.

calderon67, 2021-04-29:
> It turns out the issue had nothing to do with the display. Even with displays disabled, USB serial does not work. This makes me think Marlin was not functioning at all.
>
> The problem was actually [...]

To me this sound like the problem that I have with SKR 2. 

Do yo have any fan conected to an always running output? Because I have seen that when motherboard is not getting correctly the firmware.bin fan doesn't move.

I have to try what you said to see if it works also for me!

rhapsodyv, 2021-04-29:
The RAM issue was fixed by me. Marlin has a local copy of the board file with the fixes ram size (buildroot/share/PlatformIO/boards/).
When ram parameter is wrong, the board will not even boot.

thisiskeithb, 2021-04-29:
> The RAM issue was fixed by me. Marlin has a local copy of the board file with the fixes ram size (buildroot/share/PlatformIO/boards/).
> When ram parameter is wrong, the board will not even boot.

Marlin/PIO isn't picking it up.

rhapsodyv, 2021-04-29:
Ok, this commit renamed the boards https://github.com/MarlinFirmware/Marlin/commit/ee016e605cda97f3934b78994b390ce99ea35f04

thisiskeithb, 2021-04-29:
See https://github.com/MarlinFirmware/Marlin/pull/21741

thisiskeithb, 2021-04-30:
#21741 was merged. Closing.

closed 2021-04-30 by thisiskeithb