MarlinFirmware/Marlin #21736: [BUG] Pause printing do not moves the head to NOZZLE_PARK_POINT when ADVANCED_PAUSE_FEATURE is enabled

mriscoc, 2021-04-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When ADVANCED_PAUSE_FEATURE is enabled, pause printing do not moves the head to NOZZLE_PARK_POINT. 

### Expected behavior

Head moves to the parkpoint position on pause.

### Actual behavior

Head moves to the parkpoint position and then moves to the 0,0 on pause.

### Steps to Reproduce

1. Download the last bugfix
2. Copy the configuration files for the Ender3v2
3. In Configuration.h, enable NOZZLE_PARK_FEATURE and change the NOZZLE_PARK_POINT to `#define NOZZLE_PARK_POINT { (X_MAX_POS - 10), (Y_MAX_POS - 10), 20 }`
4. In Configuration_adv.h enable ADVANCED_PAUSE_FEATURE
5. Do a test print and pause.

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x (Apr 28 2021 20:25:42) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 V2

### Printer model

Creality Ender 3v2

### Electronics

Stock electronics

### Add-ons

none

### Your Slicer

Cura

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6395796/Configuration.zip)

fedetony, 2021-04-29:
[image]
ADVANCED_PAUSE_FEATURE Is not enabled on the attached config file.
Anyway:
If PARK_HEAD_ON_PAUSE enabled, adds the GCode M125 to pause printing and park the nozzle.
did you try M125? M0/1 keeps being just pause in my understanding.

mriscoc, 2021-04-29:
> [image]
> ADVANCED_PAUSE_FEATURE Is not enabled on the attached config file.
> Anyway:
> If PARK_HEAD_ON_PAUSE enabled, adds the GCode M125 to pause printing and park the nozzle.
> did you try M125? M0/1 keeps being just pause in my understanding.

It is my fault. I did do the test with ADVANCED_PAUSE_FEATURE enabled, and then disable it in config, I will re-upload the correct config.

fedetony, 2021-04-29:
More important is to try command M125 when is enabled. :)

mriscoc, 2021-04-29:
> More important is to try command M125 when is enabled. :)

It is enabled.

mriscoc, 2021-04-30:
Is there any confirmation of this bug?

digant73, 2021-05-04:
@mriscoc In my case it seems that REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER must be enabled in order pause properly works (nozzle parked, purge more and continue...).
If REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is disabled, when I pause a print the nozzle is moved to parking area but then it immediately moves (slowly) on top of the print and then is is paused but no purge more and continue buttons are displayed. I need to manually send a M24 gcode to resume.
It seems a bug when REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is disabled.

mriscoc, 2021-05-10:
> @mriscoc In my case it seems that REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER must be enabled in order pause properly works (nozzle parked, purge more and continue...).
> If REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is disabled, when I pause a print the nozzle is moved to parking area but then it immediately moves (slowly) on top of the print and then is is paused but no purge more and continue buttons are displayed. I need to manually send a M24 gcode to resume.
> It seems a bug when REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is disabled. You are in the same scenario reported in bug #21813

Thanks, I will try to reproduce the fix.

thisiskeithb, 2021-05-14:
With https://github.com/MarlinFirmware/Marlin/pull/21828 merged, can this be closed?

mriscoc, 2021-05-26:
> With #21828 merged, can this be closed?

Now the print head moves as expected, but there are other issues with advanced pause, but that is for other thread.

closed 2021-05-26 by mriscoc