MarlinFirmware/Marlin #21737: [BUG] (//action:pause not being sent when paused from marlin lcd)

charlespick, 2021-04-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When pressing pause on the marlin lcd while printing from a host (i use octoprint but it doesn't matter in this case) the //action:pause command isn't sent. the stop and start commands work perfectly

video: https://drive.google.com/drive/folders/1JlU5sdSv-q4osRJ-FbTiGJeFNSrrBVph?usp=sharing 

### Bug Timeline

Just discovered - tested as far back as release October 2020

### Expected behavior

Marlin should send the "//action:pause" command to the host when pause is pressed on the lcd in accordance with the RepRap gcode standard

### Actual behavior

Marlin doesn't send the pause command, leaving the lcd in a state where it can't be used and octoprint still sending commands

in the serial.log, note how a print is successfully started, canceled, and started again using host action commands. When attempting to pause as you see toward the end of the video, nothing is sent to the host and the print continues. in the serial.log the print is stopped on request of the octoprint ui at the end. 

video: https://drive.google.com/drive/folders/1JlU5sdSv-q4osRJ-FbTiGJeFNSrrBVph?usp=sharing

[serial.log](https://github.com/MarlinFirmware/Marlin/files/6395927/serial.log)

### Steps to Reproduce

1. start a print (using host start makes it more obvious that action commands as a whole are working)
2. press pause from the marlin menu
3. observe that the host is not aware of the request to pause as nothing was sent on the serial port

### Version of Marlin Firmware

Marlin bugfix-2.0.x pulled Apr 28

### Printer model

Creality CR10 V3

### Electronics

BTT SKR Mini E3 V2

### Add-ons

Bltouch

### Host Software

OctoPrint

### Additional information & file uploads

video: https://drive.google.com/drive/folders/1JlU5sdSv-q4osRJ-FbTiGJeFNSrrBVph?usp=sharing
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/6395940/configs.zip)
[serial.log](https://github.com/MarlinFirmware/Marlin/files/6395941/serial.log)
Nothing additional was printed reproducing the issue after M111 S247

thisiskeithb, 2021-04-29:
This isn't a bug. You have `HOST_PROMPT_SUPPORT` disabled in your config. Enable it and Marlin will be able to tell hosts to stop.

charlespick, 2021-04-29:
> This isn't a bug. You have `HOST_PROMPT_SUPPORT` disabled in your config. Enable it and Marlin will be able to tell hosts to stop.

why is host prompt support required to simply pause

also, after disabling advanced_pause_feature it is working

thinkyhead, 2021-05-02:
> Why is host prompt support required to simply pause?

Let's assume there is a good reason. Capice?

In this case, we don't enable many features by default because each one uses Flash and SRAM, and some boards have very little Flash and SRAM. Host Prompts are particularly string-heavy, so it's too expensive to enable by default for something like a Melzi.

closed 2021-05-02 by thinkyhead