MarlinFirmware/Marlin #21742: Reopening #21737

charlespick, 2021-04-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Reopening #21737, which was closed as a false alarm citing that not having `HOST_PROMPT_SUPPORT` enabled was the issue at hand. 

After more testing I am certain that `HOST_PROMPT_SUPPORT` being disabled is not the problem. Enabling it did not solve the problem, only adds back the `ui pause` dialog on the host as mentioned in the videos. (Those tests were done with `HOST_PROMPT_SUPPORT` enabled thus this could not be the reason)

Additionally, I have retested enabling and disabling `advanced_pause_feature` twice and reaffirm that that is what is causing the issue. After disabling it, Marlin starts sending the pause action notification again but of course that removes functionality that may be desired for other reasons and there is no reason these two features need to be exclusive of each other. 

For all other details refer to the original issue #21737

### Version of Marlin Firmware

Bugfix

thisiskeithb, 2021-04-29:
Duplicate of #21737

thisiskeithb, 2021-04-29:
Please don't open multiple reports for the same thing.

closed 2021-04-29 by thisiskeithb

charlespick, 2021-04-29:
> Please don't open multiple reports for the same thing.

What are you supposed to do instead? I replied to your comment last night immediately and it was clear from the contents of the issue that your comment made no sense and I waited more than 15 hours for you to follow up. 

Sorry if this is coming across as rude but I'm a little put off when I spent hours yesterday testing multiple versions of Marlin, multiple hosts and host devices, and multiple microcontrollers only for you to close the issue without even looking through everything I had presented.

thinkyhead, 2021-05-02:
> I waited more than 15 hours for you to follow up.

Everyone around here is just a regular Jane or Joe helping out in their spare time for fun. Some of them have jobs and lives outside of this project, so you have no right to demand a prompt response, or indeed any response. There are days and weeks at a time when I hardly have a chance to review issues due to the need for typing long paragraphs like this all day.

I don't always agree with issues being closed without a proper review either, but not everyone works like I do, with the code editor always open and ready to dive into the next problem to examine the code and see if it's really a thing.

So, I will close this issue and I'll have a closer look at your original report to see what's up.