MarlinFirmware/Marlin #21744: Error compiling on Robin Nano V3 board

galluccib, 2021-04-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Trying to compile bugfix branch for Robin Nano v3 and getting the following error...

buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../core/boards.h:376:39: error: missing binary operator before token "4220"

If I switch back to "BOARD_MKS_ROBIN_NANO" so I know platformio is good

### Bug Timeline

Today

### Expected behavior

Complete without issue

### Steps to Reproduce

Update plateformio - default_envs = mks_robin_nano_v3 (Don't plan to use USB)

Update config - #define MOTHERBOARD MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3

Clean and hit compile.

### Version of Marlin Firmware

current bugfix

### Additional information & file uploads

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/6402011/Configs.zip)

thisiskeithb, 2021-04-29:
You have the following in your config:

```cpp
// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3
#endif
```

It should look like:
```cpp
// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3
#endif
```

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-04-29 by thisiskeithb

galluccib, 2021-04-29:
> 
> 
> You have the following in your config:
> 
> ```c++
> // Choose the name from boards.h that matches your setup
> #ifndef MOTHERBOARD
>   #define MOTHERBOARD MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3
> #endif
> ```
> 
> It should look like:
> 
> ```c++
> // Choose the name from boards.h that matches your setup
> #ifndef MOTHERBOARD
>   #define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3
> #endif
> ```
> 
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:
> 
>     * [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
> 
>     * [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
> 
>     * [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group
> 
>     * [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
> 
>     * [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
> 
>     * [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server
> 
> 
> After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

yeah I noticed that a few mins... Was testing and missed it... Same error with - define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3

galluccib, 2021-04-29:
So here is something odd.... I used Marlin auto build and it worked with zero issue... Cleaned it up and hit normal compile and worked with no errors...... Very strange!!!

galluccib, 2021-04-29:
will close maybe something going on with my system..... Thanks!