MarlinFirmware/Marlin #21747: Circles printed with straight line on 12, 3, 6, 9 o'clock

Ralf-3D, 2021-04-30:
Hello.
I'm Ralf from Germany, so please excuse my english is not perfekt ;o).

Infos about my current 3D printer hardware:
- Completely new DIY designed frame with ITEM construction aluminium profiles and IGUS slide bearing technology.
- Only the stepper motors, the power supply, the heat bed, the extruder and some cables I currently use from the former ANET printer.
- 3D-Printer-Controller-Kit: RAMPS1.4 + MEGA 2560 R3 + A4988 driver for X Y Z and extruder
 --> imported from China SO NOT original Arduino.
- LCD-Display-2004-Smart-Controller_Modul
- Slicing with Prusa-Slicer and CURA, printing via RepetierHost and USB, especialy at the moment for test printing.
  Later if all things are working propperly I will print mainly from SD-Card.

### Slicer, printing host, firmware
- PrusaSlicer 2.3.0-rc3
- Cura 4.8.0
- RepetierHost V2.1.6
- alternative printing via USB with Cura 4.8.0
- alternative printing via SD-Card
- Marlin 2.0.7.2

### Did you test the latest bugfix-2.0.x (version from 26. April 2021) code?

Yes, and the problem still exists.

### Description
Full circles are printed every time with straight lines at position of 12, 3, 6 and 9 o'clock.

Foto 1:

[image]

With the thin yellow lines I highlighted the inner and outer wall of the print.
The thicker lines highlighting the brim, where you see the straight lines of the print, too.
The brim has only on SOME POSITIONS contact with the outer wall, even though I adjusted in the slicer (PrusaSlicer AND Cura) that the brim should be in direct contact with the print part without any distance.

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 

Foto 2:
[image]

While printing you can hear VERY CLEARLY that Marlin does a kind of "correction" of the g-code path on all 4 positions
where the straigt lines whithin the circle appear (!)

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 

### These are the changes, I tried to solve the problem, but ALL trials DON'T have ANY effect:
- printing via USB + RepetierHost Baud rate 250000, 128000, 115200, 76800, 38400
- printing via SD-Card
- printing several round parts from "thingiverse" in its default resolution of the .stl files
- saved or converted the self contructed 3D-part as .stl file out of my CAD software
with rough, middle, fine, extra fine resolution
- slicing with PrusaSlicer or Cura
- calibrating again the steps per mm for X, Y, Z WITH a dial indicator with a tolerance of much less 0,01 mm
- install new belts (from vendor RUTHEX)
- tension of the belts increasing AND decreasing
- compiling Marlin with Arduino IDE 1.8.12 AND 1.8.13
- printing with classic jerk
- disabeled automatic bed leveling
- printing with acceleration default values in Marlin and decreasing down to 200 mmsec²
- printing with feed rates default values in Marlin and decreasing down to 20 mm/sec

**Expected behavior:**
Printing exactly ROUND circles AND curves

**Actual behavior:**
It does NOT.

#### My files:
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/6406151/Configuration.h.zip)
[Configuration_adv.h.zip](https://github.com/MarlinFirmware/Marlin/files/6406154/Configuration_adv.h.zip)
[Test_Rundung1_fein_ni-pos-raum.STL.zip](https://github.com/MarlinFirmware/Marlin/files/6406178/Test_Rundung1_fein_ni-pos-raum.STL.zip)
[Test_Rundung1_001-1°_ni-pos-raum.STL.zip](https://github.com/MarlinFirmware/Marlin/files/6406179/Test_Rundung1_001-1._ni-pos-raum.STL.zip)
[Test_Rundung1_superfein1.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/6406194/Test_Rundung1_superfein1.gcode.zip)

### Hoping that anyone can help me to solve this problem AND to print really ROUND circles in the future ;o)

### Thank you for your HELP :o)

thisiskeithb, 2021-04-30:
Can you try printing the same thing with `CLASSIC_JERK` enabled? This will disable Junction Deviation.

There are some open issues for Junction Deviation-related moves that I suspect could be related.

XDA-Bam, 2021-05-01:
I quickly checked the gcode for obvious problems and it does not show any flat artifacts on the circles. Also, `ARC_SUPPORT` is active in the config, but not used in the example gcode.

My first suspicion would be problems with the mechanical setup or stepper motors. Did you check your driver currents / vref? Are the X and Y axis moving freely and with consistent resistance, or are they binding in certain spots? Is there any slack/play in the axes? Are there any flat spots in your belt idlers or drive gears?

thisiskeithb, 2021-05-01:
I sent `Test_Rundung1_superfein1.gcode` through [Arc Welder](https://github.com/FormerLurker/ArcWelderPlugin) to convert small segments into proper `G2`/`G3` arcs where possible: [Test_Rundung1_superfein1.aw.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/6410067/Test_Rundung1_superfein1.aw.gcode.zip)

Can you give this a try?

<details><summary>Arc-Welder Statistics:</summary>

Source: Test_Rundung1_superfein1.gcode
Lines: 28925
Size: 1.1 MB
Target: Test_Rundung1_superfein1.aw.gcode
Lines: 2424
Size: 116 kB

#### Compression
Points Compressed: 26501
Arcs Created: 669
Ratio: 9.6
Compression: 89.5%

```pre
   Min          Max     Source  Target   Change
-----------------------------------------------
  0.000mm to   0.002mm       0       0     0.0%
  0.002mm to   0.005mm       0       0     0.0%
  0.005mm to   0.010mm       0       0     0.0%
  0.010mm to   0.050mm      10       0  -100.0%
  0.050mm to   0.100mm       1       0  -100.0%
  0.100mm to   0.500mm     483     298   -38.3%
  0.500mm to   1.000mm    5427     181   -96.7%
  1.000mm to   5.000mm   21446     441   -97.9%
  5.000mm to  10.000mm       8       8     0.0%
 10.000mm to  20.000mm       8       8     0.0%
 20.000mm to  50.000mm       0     430      INF
 50.000mm to 100.000mm       0     185      INF
          >= 100.000mm       1       1     0.0%
-----------------------------------------------
       Total distance:..............30335.999mm
   Total count source:....................27384
   Total count target:.....................1552
 Total percent change:...................-94.3%
```
</details>

thinkyhead, 2021-05-02:
You might also try…
```cpp
#define MIN_STEPS_PER_SEGMENT 1
```

nextsux, 2022-01-02:
I've hit the same issue. setting MIN_STEPS_PER_SEGMENT to 1 nor using ArcWelder have not helped.

This should be a plain cylinder
[image]

Looks more like a cube with rounded corners. Infill seems to follow circular shape when printed.

I'm on custom printer, MEGA2560 + Ramps 1.6, Prusa slicer 2.3.3, Marlin FW 2.0.9.2

nextsux, 2022-01-03:
I've found my issue - it is not related to the firmware. One of my stepper broke from it's fixed position and got loose. Sorry for not checking earlier.

closed 2022-04-12 by github-actions[bot]