MarlinFirmware/Marlin #21748: [BUG] (bug summary)

Sebazzz, 2021-04-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When doing a filament load M701, and `EXTENSIBLE_UI` and `EMERGENCY_PARSER` and `HOST_PROMPT_SUPPORT` is enabled, ExtUI gets no notification to allow the filament load loop to continue. This causes 

### Bug Timeline

Started with #21671

### Expected behavior

`ExtUI::onUserConfirmRequired` is called
`wait_on_user` can be set to `false` and allows `load_filament` to exit

### Actual behavior

ExtUI is not called, an user must manually exit the loop with M108 gcode.

### Steps to Reproduce

1. Invoke M701
2. [Filament extrudes]

### Version of Marlin Firmware

2.0.8

### Printer model

Creality CR-6 SE

### Electronics

BTT SKR CR6 board; filament sensor (stock)

### Host Software

Pronterface

### Additional information & file uploads

Possible solutions:

Short-term:
- I think that `M600_PURGE_MORE_RESUMABLE` should actually be an option in `Configuration_adv.h`
- `M600_PURGE_MORE_RESUMABLE` should not be set when `EXTENSIBLE_UI`

Longer term:
- Provide `ExtUI` extensibility so that `pause_menu_response` can be set

thinkyhead, 2021-05-02:
I'm not sure how #21671 would be involved. It attempts to add functionality that didn't exist at all before. So, you can try reverting all the changes from that PR to see whether it fixes the issue. The only thing I can see that I would change is…
```diff
-   } while (TERN0(M600_PURGE_MORE_RESUMABLE, show_lcd && pause_menu_response == PAUSE_RESPONSE_EXTRUDE_MORE));
+   } while (TERN0(M600_PURGE_MORE_RESUMABLE, pause_menu_response == PAUSE_RESPONSE_EXTRUDE_MORE));
```
Since you didn't include your configurations, we have no way to test whether others of your settings may be involved in causing your issue.

Sebazzz, 2021-05-02:
> I'm not sure how #21671 would be involved. It attempts to add functionality that didn't exist at all before. 

It is, because `M600_PURGE_MORE_RESUMABLE` is set when `ADVANCED_PAUSE_FEATURE` *and* `BOTH(EMERGENCY_PARSER, HOST_PROMPT_SUPPORT)`. If you happened to have that enabled *and* an ExtUI based implementation, ExtUI does not know when a user confirmation is required, and also not able to set `pause_menu_response`.

closed 2021-07-12 by github-actions[bot]