MarlinFirmware/Marlin #21751: Filament Change

resqd1, 2021-04-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I tried the filament change setting, it doesn't give good prompts and once I did send it back to printing the extruder motor was disabled and didn't extrude at all. 

### Bug Timeline

New

### Expected behavior

Pause Printing, Keep Hotend hot, Allow change, Press resume, go back to printing

### Actual behavior

Pause printing, allowed change with a time out, then didn't resume

### Steps to Reproduce

printing, hit change filament

### Version of Marlin Firmware

Latest

### Printer model

3 v2

### Electronics

bl touch, direct drive 4.2.2 board

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

Sebazzz, 2021-04-30:
Can you reproduce with Pronterface connected? It might be the same as #21748

resqd1, 2021-04-30:
I don’t know what pronterface is. I would have to check that out. I am
New to this. Like super new lol

GOD BLESS 
Grant Parker
www.grantparkerphotography.com
Trust in the Lord with all your heart and lean not on your own understanding, in all your ways acknowledge Him and He will make your paths straight.     Proverbs 3:5-6

thinkyhead, 2021-05-02:
Duplicate of #21748.

What kind of printer is a "3 v2"?

thinkyhead, 2021-05-02:
Duplicate of #21748

closed 2021-05-02 by thinkyhead