MarlinFirmware/Marlin #21754: z auto align not working

rathorp80, 2021-04-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

on the newest release 2.0.8 with mks sgen l v2 i hit z auto align but nothing moves

### Bug Timeline

new

### Expected behavior

i expect it to do a z auto align

### Actual behavior

does nothing

### Version of Marlin Firmware

2.0.8

### Printer model

tarantula pro

### Electronics

mks sgen l v2

### Add-ons

inductive probe dual z motor

### Additional information & file uploads

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/6408312/Configuration.files.zip)

thisiskeithb, 2021-04-30:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

gerthalberg, 2021-05-25:
Having same problem. Fresh download of 2.0.8. Auto Z-Align menu item appears but nothing moves when selected.

sjasonsmith, 2021-11-25:
Given the age and lack of current information I am closing this issue.

@rathorp80 or @gerthalberg, if you still experience this problem when using the current bugfix version of Marlin, please re-open and attach your updated configuration files.

closed 2021-11-25 by sjasonsmith