MarlinFirmware/Marlin #21756: [BUG] (bug summary)

Edgar1jvega, 2021-05-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (C:\Users\admin\Downloads\Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x\Marlin\src\MarlinCore.cpp).

[image]

PLease help!!!

### Bug Timeline

idk

### Expected behavior

to work

### Actual behavior

this error

### Steps to Reproduce

that why I came here. Idk how to code

### Version of Marlin Firmware

Marlin

### Printer model

CR 10 

### Electronics

BL touch

### Add-ons

...

### Your Slicer

Cura

### Host Software

Cura

Edgar1jvega, 2021-05-01:
What does this want me to do? idk where to find all these pieces

thisiskeithb, 2021-05-01:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-05-01 by thisiskeithb

Edgar1jvega, 2021-05-01:
This is a code from here and I can't even get help? Why would this be closed? I'm sure others have this and this post could help others.

thinkyhead, 2021-05-02:
You can ignore errors from IntelliSense.