MarlinFirmware/Marlin #21757: [BUG] BLTOUCH Z_MIN CONFLICT MKS GEN L V2 + TMC2130

MaikoHermans, 2021-05-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When using a MKS Gen L V2 in the pin config there is this line:

```#define Z_DIAG_PIN                            18```
which is there for TMC2130 diag pins.

However when using this in combination with a BLTouch which uses the Z_MIN for probing the probing will continue failing.

Making a change to the ramps pins and switching Z_MIN to 19 which is actually Z_MAX the bl touch will start functioning as expected.

### Expected behavior

I expected the BLTouch to work on Z_MIN

### Actual behavior

The BLTouch didn't work when connected to Z_MIN (pin 18), tested in multiple ways with M43 S1

### Steps to Reproduce

1. Setup TMC2130's
2. Setup BLTouch on Z_MIN
3. Run M43 S1 or G28 (which will cause a 999 error)

### Version of Marlin Firmware

2.0.8

### Printer model

Anet A8

### Electronics

AM8 upgrade, genuine hiwin rails, MKS Gen L V2, Titan Aero, BLTouch

### Additional information & file uploads

[Marlin-2.0.8.zip](https://github.com/MarlinFirmware/Marlin/files/6408734/Marlin-2.0.8.zip)
zip of my full setup

ManuelMcLure, 2021-05-01:
The MKS GEN L V2.0 has jumpers that need to be removed if not using sensorless homing. This is not a bug.

MaikoHermans, 2021-05-01:
ahh hell you are right. I completely forgot about them being there...
just bounced my head against the wall for couple of hours trying to figure out why it wasn't working for no reason.

Closing this, again my bad!

closed 2021-05-01 by MaikoHermans