MarlinFirmware/Marlin #21758: [BUG] (Impossible compilation)

Balzac40, 2021-05-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Trying to compile 2.0.8 (bugfix is idle)

Error message (partial)

Compiling .pio\build\LPC1769\src\src\lcd\menu\menu_tmc.cpp.o
In file included from Marlin\src\lcd\menu\menu_temperature.cpp:31:
Marlin\src\lcd\menu\menu_temperature.cpp: In function 'void menu_temperature()':
Marlin\src\lcd\menu\menu_temperature.cpp:174:57: error: invalid conversion from 'celsius_t*' {aka 'short int*'} to 'TMenuEditItem<MenuEditItemInfo_uint16_3>::type_t*' {aka 'short unsigned int*'} [-fpermissive]  174 |       EDIT_ITEM_FAST_N(uint16_3, e, MSG_NOZZLE_STANDBY, &thermalManager.singlenozzle_temp[e], 0, thermalManager.hotend_max_target(0));
      |                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                         |
      |                                                         celsius_t* {aka short int*}
Marlin\src\lcd\menu\menu_item.h:261:39: note: in definition of macro '_MENU_INNER_P'
  261 |     MenuItem_##TYPE::action(plabel, ##V);                      \
      |                                       ^
Marlin\src\lcd\menu\menu_item.h:402:59: note: in expansion of macro '_MENU_ITEM_N_P'
  402 | #define EDIT_ITEM_FAST_N_P(TYPE, N, PLABEL, V...)         _MENU_ITEM_N_P(TYPE, N, true, PLABEL, ##V)
      |                                                           ^~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:403:55: note: in expansion of macro 'EDIT_ITEM_FAST_N_P'
  403 | #define EDIT_ITEM_FAST_N(TYPE, N, LABEL, V...)        EDIT_ITEM_FAST_N_P(TYPE, N, GET_TEXT(LABEL), ##V)
      |                                                       ^~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_temperature.cpp:174:7: note: in expansion of macro 'EDIT_ITEM_FAST_N'
  174 |       EDIT_ITEM_FAST_N(uint16_3, e, MSG_NOZZLE_STANDBY, &thermalManager.singlenozzle_temp[e], 0, thermalManager.hotend_max_target(0));
      |       ^~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:95:22: note:   initializing argument 2 of 'static void TMenuEditItem<NAME>::action(const char*, TMenuEditItem<NAME>::type_t*, TMenuEditItem<NAME>::type_t, TMenuEditItem<NAME>::type_t, screenFunc_t, bool) [with NAME = MenuEditItemInfo_uint16_3; TMenuEditItem<NAME>::type_t = short unsigned int; screenFunc_t = void (*)()]'
   95 |       type_t * const ptr,                   // Value pointer
      |       ~~~~~~~~~~~~~~~^~~
Marlin\src\lcd\menu\menu_temperature.cpp:174:57: error: invalid conversion from 'celsius_t*' {aka 'short int*'} to 'TMenuEditItem<MenuEditItemInfo_uint16_3>::type_t*' {aka 'short unsigned int*'} [-fpermissive]  174 |       EDIT_ITEM_FAST_N(uint16_3, e, MSG_NOZZLE_STANDBY, &thermalManager.singlenozzle_temp[e], 0, thermalManager.hotend_max_target(0));
      |                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                         |
      |                                                         celsius_t* {aka short int*}
Marlin\src\lcd\menu\menu_item.h:266:58: note: in definition of macro '_MENU_INNER_P'
  266 |       (encoderLine == _thisItemNr, _lcdLineNr, plabel, ##V);   \
      |                                                          ^
Marlin\src\lcd\menu\menu_item.h:402:59: note: in expansion of macro '_MENU_ITEM_N_P'
  402 | #define EDIT_ITEM_FAST_N_P(TYPE, N, PLABEL, V...)         _MENU_ITEM_N_P(TYPE, N, true, PLABEL, ##V)
      |                                                           ^~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:403:55: note: in expansion of macro 'EDIT_ITEM_FAST_N_P'
      |                                                       ^~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_temperature.cpp:174:7: note: in expansion of macro 'EDIT_ITEM_FAST_N'
  174 |       EDIT_ITEM_FAST_N(uint16_3, e, MSG_NOZZLE_STANDBY, &thermalManager.singlenozzle_temp[e], 0, thermalManager.hotend_max_target(0));
      |       ^~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:85:103: note:   initializing argument 4 of 'static void TMenuEditItem<NAME>::draw(bool, uint8_t, const char*, TMenuEditItem<NAME>::type_t*, ...) [with NAME = MenuEditItemInfo_uint16_3; uint8_t = unsigned char; TMenuEditItem<NAME>::type_t = short unsigned int]'
   85 |     FORCE_INLINE static void draw(const bool sel, const uint8_t row, PGM_P const pstr, type_t * const data, ...) {
      |                                                                                        ~~~~~~~~~~~~~~~^~~~
*** [.pio\build\LPC1769\src\src\lcd\menu\menu_temperature.cpp.o] Error 1

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6408883/Marlin.zip)

### Bug Timeline

New

### Expected behavior

Compilation success!

### Actual behavior

Compilation issue

### Steps to Reproduce

Compile 2.0.8 with my config files

### Version of Marlin Firmware

2.0.8

### Printer model

FLSUN Prusa I3+

### Electronics

BTT SKR V1.4 TURBO

### Host Software

SD Card (headless)

ellensp, 2021-05-01:
trigger is  SINGLENOZZLE_STANDBY_TEMP and  SINGLENOZZLE_STANDBY_FAN cause a different error...

ellensp, 2021-05-01:
This PR should fix it, compiles, looks good, can you test?

ellensp, 2021-05-01:
Its been committed,so should be working now.

closed 2021-05-01 by ellensp

Balzac40, 2021-05-01:
It’s OK! Thanks!

De : ellensp
Envoyé le :samedi 1 mai 2021 11:58
À : MarlinFirmware/Marlin
Cc : Balzac40; Author
Objet :Re: [MarlinFirmware/Marlin] [BUG] (Impossible compilation) (#21758)

Its been committed,so should be working now.

commit ef9ca610 (2021-05-01): Fix Singlenozzle Standby issues (#21759)

Fixes #21758
commit d00f6cbe (2021-05-02): Fix Singlenozzle Standby issues (#21759)

Fixes #21758