MarlinFirmware/Marlin #21761: [BUG] Linking error undefined reference to `ExtUI::onMeshUpdate(signed char, signed char, ExtUI::probe_state_t)'

Yuxael, 2021-05-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When compiling release-2.0.8 with configuration for Anycubic i3 Mega from Configurations repo I get error at the end:
```
Linking .pio/build/mega2560/firmware.elf
/tmp/ccqy7DTh.ltrans0.ltrans.o: In function `GcodeSuite::G26()':
<artificial>:(.text+0xbc6c): undefined reference to `ExtUI::onMeshUpdate(signed char, signed char, ExtUI::probe_state_t)'
<artificial>:(.text+0xbe36): undefined reference to `ExtUI::onMeshUpdate(signed char, signed char, ExtUI::probe_state_t)'
<artificial>:(.text+0xc1b6): undefined reference to `ExtUI::onMeshUpdate(signed char, signed char, ExtUI::probe_state_t)'
<artificial>:(.text+0xc60c): undefined reference to `ExtUI::onMeshUpdate(signed char, signed char, ExtUI::probe_state_t)'
collect2: error: ld returned 1 exit status
*** [.pio/build/mega2560/firmware.elf] Error 1
```

Same thing happens on `bugfix-2.0.x` branch.

On stock config it compiles with no issue.

### Bug Timeline

new

### Steps to Reproduce

1. Checkout Marlin 2.0.8 tag
2. Copy configs from `config/examples/AnyCubic/i3 Mega` in Configurations branch `release-2.0.8`
3. Compile using VSCode plugin Auto Build Marlin

### Version of Marlin Firmware

2.0.8

### Printer model

Anycubic i3 Mega 

### Electronics

Trigorilla 14, not v1.1

### Add-ons

TMC2209

### Your Slicer

Cura

ellensp, 2021-05-01:
Please test this PR.
It compiles, but  don't have the hardware to test

Yuxael, 2021-05-01:
Thx @ellensp. Your PR seems to fix the problem. Build it now completed without any errors.

ellensp, 2021-05-02:
This has been merged

commit 9d43570a (2021-05-02): Add missing ExtUI method (#21763)

Fixes #21761
commit 50195ec9 (2021-05-02): Add missing ExtUI method (#21763)

Fixes #21761
closed 2021-05-02 by ellensp