MarlinFirmware/Marlin #21764: [No more Serial port with MKS Bobin Nano v3]

Dtcreation, 2021-05-01:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

No more serial port :  
No usb connexion
No tft serial port active

### Expected behavior

Connexion with PC or Octoprint
Connexion with MKS TFT 28

### Actual behavior

No connexion with PC or Octoprint and no connexion with the screen

### Version of Marlin Firmware

2.0.8

### Printer model

Artillery Genius

### Electronics

MKS Robin Nano V3

### Host Software

OctoPrint

rhapsodyv, 2021-05-01:
It seems a configuration issue. You should attach your configs as requested in the template.

Dtcreation, 2021-05-01:
Not a configuration issue, same configuration than MKS Marlin

McDAK33, 2021-05-01:
I join. I have the same problem. USB works if you set default_envs = mks_robin_nano_v3 in the platformio.ini file and comment out #define USB_FLASH_DRIVE_SUPPORT in the file
Configuration_adv.h. But the connection is unstable and hangs up after a few minutes.
If you set default_envs = mks_robin_nano_v3_usb_flash_drive_msc and #define USB_FLASH_DRIVE_SUPPORT, then the board freezes when connected.

rhapsodyv, 2021-05-01:
Can you test disabling watchdog? In Configuration_adv.h comment out: `// #define USE_WATCHDOG`

McDAK33, 2021-05-01:
I tested it. When connected, the board did not hang, a COM port appeared on the PC. There is still no connection:
`Connecting...
[ERROR] Can't write to printer (disconnected?) (SerialException): Write timeout`

rhapsodyv, 2021-05-01:
> I tested it. When connected, the board did not hang, a COM port appeared on the PC. There is still no connection:
> `Connecting... [ERROR] Can't write to printer (disconnected?) (SerialException): Write timeout`

Unplug and plug the cable.

McDAK33, 2021-05-01:
Tried it - doesn't work.

rhapsodyv, 2021-05-02:
Did you try disable all serial ports? (Just to give us hint to locate the issue)

MiHeK, 2021-05-02:
Compare Configuration.h with older one. There should be #define SERIAL_PORT 1 . Instead of 0 in new version. May be this is the reason?

McDAK33, 2021-05-02:
I will do everything in my power to help.
I faced this problem for a long time #21603
MKS recommended `#define SERIAL_PORT -1` and it always worked.

McDAK33, 2021-05-02:
Not quite sure how to disable all serial ports?

Dtcreation, 2021-05-02:
I tryed with all serial desactivated. Same problem. Not a configuration.h problem. Either a platform.io or stm32 problem

ellensp, 2021-05-02:
well how about attaching your platformio.ini file then?

Dtcreation, 2021-05-02:
Not the file but the buildroot/share/platformiO, i have the Marlin original but it seems to have a bug

rhapsodyv, 2021-05-02:
Can you test this #21775?

willngton, 2021-05-02:
> Can you test this #21775?

I've tested with below PR and no longer facing the issue 

* Tested with LVGL - OK
* Tested with ColorUI - OK
* Connected via miniterm with no issues
* Did not test multi-volume
[image]

rhapsodyv, 2021-05-02:
I didn't understand your last issue.

willngton, 2021-05-02:
> I didn't understand your last issue.

Upload a video sample for my Robin Nano V3.0

### With usb stick connected wont come up till re-seat button is pressed 
https://user-images.githubusercontent.com/5961905/116819335-b8665100-ab3d-11eb-9f5d-d57e8d507c0d.mp4

https://user-images.githubusercontent.com/5961905/116819347-bd2b0500-ab3d-11eb-8a21-b2ed5f3e4ae5.mp4

McDAK33, 2021-05-02:
Tested with MKS H43. Results:
- COM port is present
- USB disk present
- The connection is established, but when you try to print a test cube, the connection is disconnected at the sixth minute of printing. After using `M575 B250`, I managed to print the cube to the end.
- The temperature on the H43 display is not correct (as a reminder).

I conclude that the connection has appeared, but it is unstable.
Some problems with the connection speed.

Dtcreation, 2021-05-02:
HI,

With the PR everything seems good to me, but i have to test a long print to confirm

Thanks

rhapsodyv, 2021-05-02:
> Tested with MKS H43. Results:
>
> * COM port is present
> * USB disk present
> * The connection is established, but when you try to print a test cube, the connection is disconnected at the sixth minute of [...]

It’s seems an issue with H43, not the nano v3.

McDAK33, 2021-05-02:
> It’s seems an issue with H43, not the nano v3.

H43 has no effect on problems with Robin Nano V3. I also have MKS TS35 V2.0. For a long time there was a problem with the USB connection, and this screen was tested.
I did not notice any difference in the behavior of the Nano V3.

rhapsodyv, 2021-05-02:
@McDAK33  All `M575` does is the same that is done at boot time: start serial with `BAUDRATE` from the `Configuration.h`. Maybe what is "fixing" your issue is that it restart the serial. But none of this code is exclusive to Nano V3.

Where do you connect the H43 in the nano v3 board?

rhapsodyv, 2021-05-02:
> ### With usb stick connected wont come up till re-seat button is pressed
>  NanoV3_Boot_USB.mp4 
> 
>  NanoV3_Boot_NoUSB.mp4

I tested and I don't have this issue. Can you attach your configs?

McDAK33, 2021-05-02:
@rhapsodyv I understand that perfectly. `Configuration.h` is always set to` #define BAUDRATE 250000`. But the fact remains. After switching on, the connection is interrupted, and after `M575`, the print goes to the end. Why is this happening and I would like to find out.

rhapsodyv, 2021-05-02:
> @rhapsodyv I understand that perfectly. `Configuration.h` is always set to` #define BAUDRATE 250000`. But the fact remains. After switching on, the connection is interrupted, and after `M575`, the print goes to the end. Why is this happening and I would like to find out.

Sure, for some reason the serial is freezing,  and when you send `M575` it's restarting it. Where do you connect the H43 in the nano v3?

McDAK33, 2021-05-02:
[image]

willngton, 2021-05-02:
> I tested and I don't have this issue. Can you attach your configs?

[Configuration_w.h.txt](https://github.com/MarlinFirmware/Marlin/files/6411963/Configuration_w.h.txt)
[Configuration_adv_w.h.txt](https://github.com/MarlinFirmware/Marlin/files/6411964/Configuration_adv_w.h.txt)

rhapsodyv, 2021-05-02:
> > I tested and I don't have this issue. Can you attach your configs?
> 
> [Configuration_w.h.txt](https://github.com/MarlinFirmware/Marlin/files/6411963/Configuration_w.h.txt)
> [Configuration_adv_w.h.txt](https://github.com/MarlinFirmware/Marlin/files/6411964/Configuration_adv_w.h.txt)

Are you using external power in your tests? Or usb power?

willngton, 2021-05-02:
> Are you using external power in your tests? Or usb power?

I am using a 15A 12v external power supply

rhapsodyv, 2021-05-04:
So, this can be closed?

The H43 issue is being handle in #21603, right?

PR #21775 (2021-05-02): Multi Volume Followup 1
### Description

Fix a type (wrong) call in the multi volume PR

### Benefits

May fix #21764 

### Related Issues

#21344
closed 2021-05-04 by Dtcreation