MarlinFirmware/Marlin #21766: [BUG] Z2 axis moving faster then Z, SKR v1.3 Dual Steppers

thezak48, 2021-05-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Trying to get Dual Z steppers to work, however it seems that no mater what I do Z2 seems to move faster then Z.

If I use a y harness for the steppers and use one driver the axis run perfectly fine no issues

I have set:
#define NUM_Z_STEPPER_DRIVERS 2
#define Z2_DRIVER_TYPE TMC2209
and changed the current for Z2

I have even attempted setting E1 with the above setting and nothing. Config files attached

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/6409961/Config.zip)

### Expected behavior

I expect both Z axis to move at the same rate

### Actual behavior

Z2 moves faster then Z

### Version of Marlin Firmware

Marlin bugfix-2.0.x

### Printer model

Ender 3

### Electronics

SKR v1.3, TMC2209

### Your Slicer

Simplify3D

### Host Software

OctoPrint

thisiskeithb, 2021-05-02:
Some tings to check/try:

- All jumpers are removed from under the drivers (since they are controlled via UART)
- UART jumpers are installed for each driver
- DIAG jumpers for both Z & E1 are removed

Also, what does `M122` report?

ellensp, 2021-05-02:
Also is this a new seconds Z steppr motor, any chance its a 0.9 degree when the original is a 1.8 degree stepper?  or vice versa

ellensp, 2021-05-02:
I just tried this. Using your configs same hardware (skr 1,3, TMC2209's, Dual Z plugged into Z and E1, cr10 lcd and two known identical stepper drivers.
Since you left steps to reproduce the issue blank. I used the LCD, Move Z 10 mm. AN used gcodes G1 Z{numbers} 

It woks as expected.

video of the test https://photos.app.goo.gl/f1eBHYewQ2e3U3W2A

It has to be a hardware issue on your setup.

closed 2021-05-02 by ellensp