MarlinFirmware/Marlin #21767: [BUG] 2.0.8 "Probing Failed"

rq3, 2021-05-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Compiled 2.0.8 using the latest configuration/examples/delta/Anycubic/Predator configuration.h and configuration_adv.h
Compile went flawlessly, and bin file loaded fine. The printer prints. BUT, any attempt to use the probe for delta calibration, M48 repeatability, or bed leveling immediately results in a "Probing Failed" message after the effector lowers to the deploy/stow position.

### Bug Timeline

New bug which showed up with 2.0.8. Marlin 2.0.7.2 works fine.

### Expected behavior

I expect the machine to succeed with an M48 repeatability test, a delta calibration, or bed leveling.

### Actual behavior

The machine goes to the deploy/stow position, requests a deploy, requests a stow, and then reports "Probing Failed".

### Steps to Reproduce

1) Download Marlin 2.0.8
2) Download the Anycubic Predator configuration files for 2.0.8
3) Modify platformio.ini for the trigorilla_pro environment
4) Successfully compile 2.0.8 using Visual Studio
5) Install the resulting firmware.bin on the printer
6) Initialize the printer EEPROM
7) Power cycle the printer
8) Attempt to use the z axis probe in any way

### Version of Marlin Firmware

2.0.8

### Printer model

Anycubic Predator

### Electronics

stock Chitu Trigorilla Pro

### Add-ons

none

### Your Slicer

Cura

thisiskeithb, 2021-05-01:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

rq3, 2021-05-01:
Here you go. Copied from the log in Repetier Host:

[2.0.8_Probing_Failure.txt](https://github.com/MarlinFirmware/Marlin/files/6410288/2.0.8_Probing_Failure.txt)

lujios, 2021-05-02:
I think I have located the bug. There is a recent modification in motion.cpp, which causes the mentioned error. If you think it is convenient, I will open a pull request, with the solution that I propose. 
The solution is tested, with OK result.

rq3, 2021-05-02:
I just compiled and installed the very latest bugfix (May 2 AM) with its associated example configuration files. It no longer says "Probing Failed". Instead it moves to the deploy/stow position and asks for a deploy. I did so. It then immediately says "Stow Probe". I did so. It then immediately says "Deploy Probe". This goes on, apparently, forever. The effector never moved after the first positioning.

rhapsodyv, 2021-05-02:
Can you try reverting this commit https://github.com/MarlinFirmware/Marlin/commit/4044ed8783c71dd66d64737173ca31c8a0ea0f5a and test if the issue is gone?

lujios, 2021-05-02:
I am sorry.
My setup is ready for SENSORLESS_PROBING. It is very possible that for this reason it will not work if you have probe_Z.
Have you tried with your config files and modifying motion.cpp? It is not a configuration problem, if it was working before.
My pull request, it was not yet merged into bugfix_2.0.x. My pull request was not yet merged into bugfix_2.0.x, so you didn't use my fix.
Thank.

rq3, 2021-05-02:
> Can you try reverting this commit 4044ed8
> and test if the issue is gone?

That works. I downloaded a fresh copy of the latest bugfix, replaced the 
motion.cpp file, and used the configuration files from my previous 
Marlin 2.0.8 installation. M48 works fine, and delta calibration appears 
to be doing its thing, although I did not let it go to completion.

Would you like me to replace the motion.cpp file in my Marlin 2.0.8 and 
see what happens?

Best Regards,
Rip

rq3, 2021-05-02:
Victor, I replaced the motion.cpp in my previously compiled copy of 
2.0.8 with the motion.cpp commit 4044ed8, recompiled, and installed. It 
seems to work fine for M48 and delta calibration. I haven't tried a 
complete bed leveling yet, but seems to start the process with no 
issues, so no reason why it shouldn't work there as well.

Thanks,
Rip

lujios, 2021-05-02:
> Eso funciona. Descargué una copia nueva de la última corrección de errores, reemplacé el archivo motion.cpp y usé los archivos de configuración de mi instalación anterior de Marlin 2.0.8. M48 funciona bien, y la calibración delta parece estar haciendo lo suyo, aunque no la dejé completar. ¿Le gustaría que reemplace el archivo motion.cpp en mi Marlin 2.0.8 y ver qué sucede? Saludos cordiales, Rip
> [...](#)

I am glad to hear. If possible, I appreciate it. So I would test my solution with more configurations.

lujios, 2021-05-02:
> That works. I downloaded a fresh copy of the latest bugfix, replaced the motion.cpp file, and used the configuration files from my previous Marlin 2.0.8 installation. M48 works fine, and delta [...]

I am glad to hear. If possible, I appreciate it. So I would test my solution with more configurations.

rq3, 2021-05-02:
I should note that my delta has two z axis probes. One is the stock Anycubic microswitch, and the other is a prototype piezo sensor under the bed. I can switch between the two, and they both work with this motion.cpp change (neither would work before in 2.0.8).

Both probes trigger the same z-min pin on the Trigorilla Pro board.

It would appear that motion.cpp commit 4044ed8 is pretty robust, at least for this board.

rhapsodyv, 2021-05-02:
> It would appear that motion.cpp commit [4044ed8](https://github.com/MarlinFirmware/Marlin/commit/4044ed8783c71dd66d64737173ca31c8a0ea0f5a) is pretty robust, at least for this board. I think that if this commit can be merged into 2.0.8 you can close this issue.

There're some confusion here. I think https://github.com/MarlinFirmware/Marlin/commit/4044ed8783c71dd66d64737173ca31c8a0ea0f5a added the bug by mistake.

rhapsodyv, 2021-05-02:
@rq3 can you test this PR https://github.com/MarlinFirmware/Marlin/pull/21781 ?

It fixes a typo in the code that introduced the issue.

rq3, 2021-05-02:
Yes, that works. All probing functions work with that simple change in 
motion.cpp. As a retired radio frequency hardware engineer, for the life 
of me I don't know how you software folks do this.

Best Regards,
Rip

rq3, 2021-05-02:
Just for completeness, you should know I made the change directly in the 
motion.cpp file in my copy of Marlin 2.0.8

rhapsodyv, 2021-05-02:
Now that you confirmed the fix in #21781  works, @thinkyhead may merge it right away.

rq3, 2021-05-02:
@rhapsodyv , sounds good. At least it solved all the issues I can see on my machine.

rhapsodyv, 2021-05-02:
Merged!

PR #21781 (2021-05-02): Blocking Move Followup 2 (fix delta failed probe)
### Description

Fix a typo in 3517fbdcbdbb3ba790b0aac55e3ffb03e94f3307

I follows the hint of @lujios in #21774 

Co-author: @lujios 

### Benefits

Fix #21767 
Fix #21620
closed 2021-05-02 by rhapsodyv