MarlinFirmware/Marlin #21769: [BUG] Linear Advance makes the extruder to loud to use.

PigeonFX, 2021-05-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Enabling `LIN_ADVANCE` makes the extruder extremely loud compared to it being disabled (I'm talking louder than a4988 loud).  And seems to be most pronounced on retraction (2mm @25mm/s).

I have tried a different stepper motor and switched drivers and even extruders from a BMG to a MK8 stock style.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6410677/Configuration.zip)

### Bug Timeline

Started happing after updating to 2.0.7.2 and using LA for the first time.

### Expected behavior

I expected my extruder to make more noise due to the constant movement to regulate pressure, but just not to this extent. 

### Actual behavior

Very load, especially on retractions. 

### Steps to Reproduce

Change K to anything other then 0 than

### Version of Marlin Firmware

2.0.7.2

### Printer model

Ender 3

### Electronics

SKR E3 Mini V2

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

swilkens, 2021-05-02:
This is likely a known issue;
https://github.com/MarlinFirmware/Marlin/issues/10272

You might try to enable `#define LA_DEBUG` in `Configuration_adv.h` - i expect it's throwing out "max accell" notices. If so, you may try to increase your E-jerk.

There has been discussion about looking over JD and comparing with pressure advance from Klipper, but so far those efforts have not been taken afaik.

PigeonFX, 2021-05-02:
Thank you, will have a look at that link, and enable `LA_DEBUG`. 

Have a pi with Klipper that is slowly making it's way towards this printer, Love Marlin but love being able to print without hearing protection more!

PigeonFX, 2021-05-02:
Have enabled `LA_DEBUG` But, don't seem to be getting anything out of the normal via the terminal (I'm using the terminal in Octoprint, if that matters?)

```
#define LIN_ADVANCE
#if ENABLED(LIN_ADVANCE)
  //#define EXTRA_LIN_ADVANCE_K // Enable for second linear advance constants
  #define LIN_ADVANCE_K 0.44    // Unit: mm compression per 1mm/s extruder speed
  #define LA_DEBUG            // If enabled, this will generate debug information output over USB.
  #define EXPERIMENTAL_SCURVE   // Enable this option to permit S-Curve Acceleration
#endif
```

swilkens, 2021-05-05:
Does the noise lever reduce when you use lower K-values? It may well be that the friction in the bowden system is too much, causing missed steps

Do you use retraction and, if so, do you notice missed steps there as well?

PigeonFX, 2021-05-05:
> Does the noise lever reduce when you use lower K-values? It may well be that the friction in the bowden system is too much, causing missed steps

No, tried halving the value and it was the same horrible noise, only way to get it gone is a 0 value. 

> Do you use retraction and, if so, do you notice missed steps there as well?

tried with 1 and 2mm retraction not missed steps, and even did a sanity check for hardware and installed Klipper and configured Pressure advance, and it was near silent.

swilkens, 2021-05-05:
Is the noise the same on 2.0.7.2 release?

If so, then it's likely inherent to the way we implement LA.

@Sebastianv650

PigeonFX, 2021-05-05:
> Is the noise the same on 2.0.7.2 release?

Good question, think it was upgraded from 2.0.2 and not sure the stock config had LA enabled, will roll back to 2.0.7.2 and test when I get the time.

swilkens, 2021-05-05:
I took a look at the klipper implementation and can confirm it's rather more silent compared to current 2.0.8. I expect you will find he same in 2.0.7.2. If so, this is indeed a known issue.

PigeonFX, 2021-05-06:
2.0.7.2 tested and unfortunately the same issue. 

So, guess its the known issue, and off to Klipper till v2.?

sarvenn, 2021-05-07:
Not only it makes noise but also there is significant print quality degredation.

iZz0s, 2021-05-18:
ive tried today klipper, too.

pressure advance is at klipper fully silent, on marlin i got noise like with old A4988. anything new about that bug?

thierryzoller, 2021-05-27:
Duplicate of #10272 ?

sarvenn, 2021-07-27:
Is it possible to change LA algorithm to something similar to Klipper's PA feature?

I can not use LA due to degradaded print quality and abnormal noises.
Is there any document or video how to solve this? or how to configure marlin to use LA (LA + classic jerk or LA + JD, etc.)

Kelduum, 2021-07-27:
Possibly useful for diagnostic purposes - I swapped my extruder stepper from the sock Ender 3 (Pro) stepper to a basic pancake stepper a few weeks back after moving to a BMG extruder, and immediately noticed the noise when using linear advance disappeared entirely.

Swapping it back without making any changes resulted in the stepper noise again, without changing anything else.

Possibly also helpful, but the tone generated by the stock Ender 3 stepper with LA enabled seems to change the frequency of the tone based on the LA values.

If it helps, board is a SKR Mini E3 V2.

thinkyhead, 2022-04-21:
> Is it possible to change LA algorithm to something similar to Klipper's PA feature?

If we can find a smart programmer to implement it.

thisiskeithb, 2022-08-04:
Linear Advance should be quieter / improved after https://github.com/MarlinFirmware/Marlin/pull/24533

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

closed 2022-09-29 by thisiskeithb