MarlinFirmware/Marlin #21772: 2.0.8 Missing pin config in GTR v1.0

DJDOUBLEG, 2021-05-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Many error when compiling marlin with VSC.
from Marlin\src\HAL\STM32\tft\tft_spi.cpp:24:
Marlin\src\HAL\STM32\tft\tft_spi.h: In static member function 'static void TFT_SPI::DataTransferEnd()':
Marlin\src\HAL\STM32\tft\tft_spi.h:59:41: error: 'TFT_CS_PIN' was not declared in this scope; did you mean 'TFT_SPI'?
   59 |   static void DataTransferEnd() { WRITE(TFT_CS_PIN, HIGH); };
      |                                         ^~~~~~~~~~
C:\Users\User\.platformio\packages\framework-arduinoststm32@src-c956ec6dd219e931e12b035aaba6083c\cores\arduino\stm32/PinNamesTypes.h:129:34: note: in definition of macro 
'STM_PORT'
  129 | #define STM_PORT(X) (((uint32_t)(X) >> 4) & 0xF)
Compiling .pio\build\BIGTREE_GTR_V1_0_usb_flash_drive\src\src\HAL\shared\Delay.cpp.o
      |                                  ^
Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/fastio.h:58:49: note: in expansion of macro 'digitalPinToPinName'
   58 |   #define _WRITE(IO, V) (FastIOPortMap[STM_PORT(digitalPinToPinName(IO))]->BSRR = _BV32(STM_PIN(digitalPinToPinName(IO)) + ((V) ? 0 : 16)))
      |                                                 ^~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/fastio.h:69:33: note: in expansion of macro '_WRITE'
   69 | #define WRITE(IO,V)             _WRITE(IO,V)
      |                                 ^~~~~~
Marlin\src\HAL\STM32\tft\tft_spi.h:59:35: note: in expansion of macro 'WRITE'
   59 |   static void DataTransferEnd() { WRITE(TFT_CS_PIN, HIGH); };
      |                                   ^~~~~
Marlin\src\HAL\STM32\tft\tft_spi.h: In static member function 'static void TFT_SPI::WriteReg(uint16_t)':
Marlin\src\HAL\STM32\tft\tft_spi.h:63:46: error: 'TFT_A0_PIN' was not declared in this scope; did you mean 'TFT_SPI'?
   63 |   static void WriteReg(uint16_t Reg) { WRITE(TFT_A0_PIN, LOW); Transmit(Reg); WRITE(TFT_A0_PIN, HIGH); }
      |                                              ^~~~~~~Compiling .pio\build\BIGTREE_GTR_V1_0_usb_flash_drive\src\src\HAL\shared\HAL_MinSerial.cpp.o
~~~
C:\Users\User\.platformio\packages\framework-arduinoststm32@src-c956ec6dd219e931e12b035aaba6083c\cores\arduino\stm32/PinNamesTypes.h:129:34: note: in definition of macro 
'STM_PORT'
  129 | #define STM_PORT(X) (((uint32_t)(X) >> 4) & 0xF)
      |                                  ^
Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/fastio.h:58:49: note: in expansion of macro 'digitalPinToPinName'
Compiling .pio\build\BIGTREE_GTR_V1_0_usb_flash_drive\src\src\HAL\shared\backtrace\backtrace.cpp.o
   58 |   #define _WRITE(IO, V) (FastIOPortMap[STM_PORT(digitalPinToPinName(IO))]->BSRR = _BV32(STM_PIN(digitalPinToPinName(IO)) + ((V) ? 0 : 16)))
      |                                                 ^~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/fastio.h:69:33: note: in expansion of macro '_WRITE'
   69 | #define WRITE(IO,V)             _WRITE(IO,V)
      |                                 ^~~~~~
Marlin\src\HAL\STM32\tft\tft_spi.h:63:40: note: in expansion of macro 'WRITE'
   63 |   static void WriteReg(uint16_t Reg) { WRITE(TFT_A0_PIN, LOW); Transmit(Reg); WRITE(TFT_A0_PIN, HIGH); }
      |                                        ^~~~~
In file included from Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/HAL.h:31,
                 from Marlin\src\HAL\STM32\tft\../../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\STM32\tft\../../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32\tft\tft_spi.cpp:24:
Marlin\src\HAL\STM32\tft\tft_spi.cpp: In static member function 'static void TFT_SPI::Init()':
Marlin\src\HAL\STM32\tft\tft_spi.cpp:37:13: error: 'TFT_A0_PIN' was not declared in this scope; did you mean 'TFT_SPI'?
   37 |   OUT_WRITE(TFT_A0_PIN, HIGH);
      |             ^~~~~~~~~~
Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/fastio.h:66:41: note: in definition of macro '_SET_OUTPUT'
   66 | #define _SET_OUTPUT(IO)         pinMode(IO, OUTPUT)                               //!< Output Push Pull Mode & GPIO_NOPULL
      |                                         ^~
Marlin\src\HAL\STM32\tft\tft_spi.cpp:37:3: note: in expansion of macro 'OUT_WRITE'
   37 |   OUT_WRITE(TFT_A0_PIN, HIGH);
      |   ^~~~~~~~~
Marlin\src\HAL\STM32\tft\tft_spi.cpp:38:13: error: 'TFT_CS_PIN' was not declared in this scope; did you mean 'TFT_SPI'?
   38 |   OUT_WRITE(TFT_CS_PIN, HIGH);
      |             ^~~~~~~~~~
Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/fastio.h:66:41: note: in definition of macro '_SET_OUTPUT'
   66 | #define _SET_OUTPUT(IO)         pinMode(IO, OUTPUT)                               //!< Output Push Pull Mode & GPIO_NOPULL
      |                                         ^~
Marlin\src\HAL\STM32\tft\tft_spi.cpp:38:3: note: in expansion of macro 'OUT_WRITE'
   38 |   OUT_WRITE(TFT_CS_PIN, HIGH);
      |   ^~~~~~~~~
In file included from C:\Users\User\.platformio\packages\framework-arduinoststm32@src-c956ec6dd219e931e12b035aaba6083c\cores\arduino/Arduino.h:56,
                 from Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/../shared/Marduino.h:36,
                 from Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/HAL.h:28,
                 from Marlin\src\HAL\STM32\tft\../../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\STM32\tft\../../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32\tft\tft_spi.cpp:24:
Marlin\src\HAL\STM32\tft\tft_spi.cpp:40:75: error: 'TFT_SCK_PIN' was not declared in this scope; did you mean 'SD_SCK_PIN'?
   40 |   if ((spiInstance = (SPI_TypeDef *)pinmap_peripheral(digitalPinToPinName(TFT_SCK_PIN),  PinMap_SPI_SCLK)) == NP) return;
      |                                                                           ^~~~~~~~~~~
C:\Users\User\.platformio\packages\framework-arduinoststm32@src-c956ec6dd219e931e12b035aaba6083c\cores\arduino/pins_arduino.h:249:49: note: in definition of macro 'digitalPinToPinName'
  249 | #define digitalPinToPinName(p)      (((uint32_t)p < NUM_DIGITAL_PINS) ? digitalPin[p] : NC)
      |                                                 ^
compilation terminated due to -fmax-errors=5.
*** [.pio\build\BIGTREE_GTR_V1_0_usb_flash_drive\src\src\HAL\STM32\tft\tft_spi.cpp.o] Error 1

### Bug Timeline

Old

### Expected behavior

Who will be fixed

### Actual behavior

The compiling stops with an error code

### Steps to Reproduce

1. enable GTR
2. enable tft generic
3. compiling

### Version of Marlin Firmware

2.0.8

### Printer model

any

### Electronics

BTT GTR

### Additional information & file uploads

Missing this lines in pins file:
#define TFT_CS_PIN                     PB12
      #define TFT_A0_PIN                     PB10
      #define TFT_SCK_PIN                    PB13
      #define TFT_MISO_PIN                   PB14
      #define TFT_MOSI_PIN                   PB15

      #define TOUCH_INT_PIN                  PG6
      #define TOUCH_MISO_PIN                 PG1
      #define TOUCH_MOSI_PIN                 PC10
      #define TOUCH_SCK_PIN                  PG8
      #define TOUCH_CS_PIN                   PA8
      #define BTN_EN1                        PD10
      #define BTN_EN2                        PH10

thinkyhead, 2021-05-02:
Is that all? Most pins files that include these defines also wrap them up with…
```cpp
#if ANY(TFT_COLOR_UI, TFT_LVGL_UI, TFT_CLASSIC_UI)
```
… and then they usually also include some fallback `TOUCH_CALIBRATION_*` values. Does Geeetech have a completed pins file that I could look at for comparison?

thinkyhead, 2021-05-02:
I suspect that the `TOUCH_MISO_PIN` is incorrect, and should be `PG7`.

rhapsodyv, 2021-05-02:
BTT has no EXP compatible TFT. The user can only use a custom TFT.

is this the case??

The TFT pins config wasn’t added on GTR yet because until now no user tried to use a custom TFT with a GTR board...

Note: BTT TFT35 (and so on) are serial TFT, and they don’t use that TFT generic config...

DJDOUBLEG, 2021-05-02:
> BTT has no EXP compatible TFT. The user can only use a custom TFT.
> 
> is this the case??
> 
> The TFT pins config wasn’t added on GTR yet because until now no user tried to use a custom TFT with a GTR board...
> 
> Note: BTT TFT35 (and so on) are serial TFT, and they don’t use that TFT generic config...

Are you absolutely sure of that?

[image]
[image]

rhapsodyv, 2021-05-02:
😮 
I didn't know that. Is it already released?!

DJDOUBLEG, 2021-05-02:
> 
> 
> BTT has no EXP compatible TFT. The user can only use a custom TFT.
> 
> is this the case??
> 
> The TFT pins config wasn’t added on GTR yet because until now no user tried to use a custom TFT with a GTR board...
> 
> Note: BTT TFT35 (and so on) are serial TFT, and they don’t use that TFT generic config...

And why include this line in SKR 1.4 pins file?
[image]

DJDOUBLEG, 2021-05-02:
> 
> 
> 😮
> I didn't know that. Is it already released?!

Not now :D

rhapsodyv, 2021-05-02:
I added it to the SKR 1.4 turbo while testing the MKS-TS35 on it. It was just to add TFT support for LPC.

DJDOUBLEG, 2021-05-02:
Its works with my GTR :)
[image]

thisiskeithb, 2021-05-02:
> 😮
> I didn't know that. Is it already released?!

That’s because it’s a beta product still being tested and shouldn’t have been posted here 🙂

I think we can still add the pins for these displays though.

DJDOUBLEG, 2021-05-02:
> 
> 
> > 😮
> > I didn't know that. Is it already released?!
> 
> That’s because it’s a beta product still being tested and shouldn’t have been posted here 🙂
> 
> I think we can still add the pins for these displays though.

removed the photo :)

thisiskeithb, 2021-05-02:
> removed the photo :)

Too late now 🙂 

For this new TFT from BigTreeTech, we've been using `BTT_TFT35_SPIV1_0`which sets the following:

```cpp
#if ENABLED(BTT_TFT35_SPIV1_0)
  #define TFT_COLOR_UI                 // Classic UI works too
  #define TFT_ROTATION TFT_ROTATE_180
  #define TOUCH_SCREEN
#endif
```

rhapsodyv, 2021-05-02:
They should add this TFT to this section: https://github.com/MarlinFirmware/Marlin/blob/64a7dfbe7d76a4d05663e597fbaeeecb95069a66/Marlin/src/inc/Conditionals_LCD.h#L1069

thisiskeithb, 2021-05-02:
> They should add this TFT to this section:
> 
> https://github.com/MarlinFirmware/Marlin/blob/64a7dfbe7d76a4d05663e597fbaeeecb95069a66/Marlin/src/inc/Conditionals_LCD.h#L1069

Yes, there’s support pending. Like I said, this FR/issue should not have been posted here in the first place since it’s beta hardware/still being tested.

thisiskeithb, 2021-05-02:
Gábor: Please reach out to BigTreeTech directly for the status of this TFT. I'm going to close this FR for now since this hardware is not available to the general public and is still being beta tested.

closed 2021-05-02 by thisiskeithb

DJDOUBLEG, 2021-05-02:
> 
> 
> > removed the photo :)
> 
> Too late now 🙂
> 
> For this new TFT from BigTreeTech, the beta testers have been using `BTT_TFT35_SPIV1_0`which sets the following:
> 
> ```c++
> #if ENABLED(BTT_TFT35_SPIV1_0)
>   #define TFT_COLOR_UI                 // Classic UI works too
>   #define TFT_ROTATION TFT_ROTATE_180
>   #define TOUCH_SCREEN
> #endif
> ```
> 
> Since this TFT is being beta tested by a small group of users, support should really come from BigTreeTech as it's not a bug in Marlin and I'm not even sure if this hardware is final yet.

This settigns is not enough!

thisiskeithb, 2021-05-02:
> This settigns is not enough!

I know there is more code changes required, this was just some of them to setup the TFT.

This TFT is not available for sale.

You're a beta test for BigTreeTech and should know better than to post here.

DJDOUBLEG, 2021-05-02:
Not need any only the pins file missig some lines.

thinkyhead, 2021-05-02:
Ok. So, they deleted their GitHub account…. So I will hold my questions for another time.