MarlinFirmware/Marlin #21777: src/HAL/STM32/timers.cpp includes file that doesn't exist

LeslieTroyer, 2021-05-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When using stm32 architecture and software serial to communicate with the TMC's the file 
../Marlin/src/HAL/STM32/timers.cpp includes <SoftwareSerial.h>  This file existed at version 2.0.6 but not in 2.0.8.  Lack of this file causes the build to fail.

Marlin/src/HAL/STM32/timers.cpp:47:12: fatal error: SoftwareSerial.h: No such file or directory
   47 |   #include <SoftwareSerial.h>
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/BTT_SKR_SE_BX/src/src/HAL/STM32/timers.cpp.o] Error 1

#if HAS_TMC_SW_SERIAL
  #include <SoftwareSerial.h>
  #ifndef SWSERIAL_TIMER_IRQ_PRIO
    #define SWSERIAL_TIMER_IRQ_PRIO SWSERIAL_TIMER_IRQ_PRIO_DEFAULT
  #endif
#endif

I copied SoftwareSerial.cpp & SoftwareSerial.h from 2.0.6 and that failed to compile due to other changes in the code.

### Bug Timeline

2.0.8

### Expected behavior

expected it to compile

### Actual behavior

compile failed

### Steps to Reproduce

build with BTT_SKR_SE_BX board

### Version of Marlin Firmware

2.0.8

### Printer model

BIQU BX

### Electronics

Stock

### Add-ons

none

### Host Software

OctoPrint

### Additional information & file uploads

[config's.zip](https://github.com/MarlinFirmware/Marlin/files/6411552/config.s.zip)

ellensp, 2021-05-02:
Cannot reproduce this error. There are also errors in the provided Configs
`#define Z_PROBE_FEEDRATE_SLOW (Z_PROBE_SPEED_FAST / 2)`
but its now called 
`#define Z_PROBE_FEEDRATE_SLOW (Z_PROBE_FEEDRATE_FAST / 2)`

Is #define EXTENSIBLE_UI valid for this type of LCD (TFT_1024x600_LTDC)  ?
As it won't compile with this enabled, but compiles is fine with BTT_STYLE_UI.

rhapsodyv, 2021-05-02:
@ellensp 

I didn’t review BTT recent ltdc PR, but extensible ui is used only by serial tft, as far I know...
So, in my view you are right.

And SoftwareSerial ships with stm32duino, so if there’s an issue, is with the env. But if you cannot reproduce, probably is an user environment issue.

thisiskeithb, 2021-05-02:
`BTT_STYLE_UI` hasn't been ported upstream yet. You should use `TFT_CLASSIC_UI` or `TFT_COLOR_UI` with the Biqu BX.

ericsnis, 2021-06-10:
Similar problem on the SKR Mini E3 v1.2 (STM32F103RC_btt_512K target).  I cherry picked commits up to Creality v4 with STM32 HAL #21999.  Things compile correctly if I leave out that commit, but something in that commit breaks timers.
PlatformIO 5.1.1 was cleanly installed and it automatically grabbed the STM32 12.1.1 platform.

[configuration SKR mini E3 v1.2.zip](https://github.com/MarlinFirmware/Marlin/files/6627454/configuration.SKR.mini.E3.v1.2.zip)

ellensp, 2021-06-10:
@ericsnis your configs upgraded to bugfix compiles without issues on bugfix.

rhapsodyv, 2021-06-10:
SoftwareSerial ships with stm32duino. It's impossible to not find it.. 

It's seems just a corrupted package install... 

Try delete the folder `~/.platformio/packages` (.platformio in your home folder) and `.pio` inside marlin folder.

This will make pio download the libraries again.

closed 2021-06-22 by thisiskeithb