MarlinFirmware/Marlin #21778: [BUG] Garbled display after update from 2.0.7.2 to 2.0.8

hartmannathan, 2021-05-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After updating from 2.0.7.2 to 2.0.8, the display is garbled - see images. Downgrading back to 2.0.7.2 fixes the problem. Problem exists in bugfix-2.0.x branch too. Hardware: BTT GTR V1.0, REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.

I'll be happy to bisect from 2.0.7.2 to 2.0.8 to find the commit that broke it, if someone could tell me the git incantation :-)

### Working correctly with Marlin 2.0.7.2:

[image]

### Garbled with Marlin 2.0.8:

[image]

### Bug Timeline

Appears in 2.0.8.

### Version of Marlin Firmware

2.0.8

### Printer model

Creality CR-10

### Electronics

BTT GTR V1.0 and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

### Your Slicer

Other (explain below)

### Host Software

Other (explain below)

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6411726/Configuration.zip)

ellensp, 2021-05-02:
several things changes that effecting timing,and it was found that several platforms had incorrect timings to start with. Now corrected. But it means that tings like this in your  Marlin/src/pins/stm32f4/pins_BTT_GTR_V1_0.h file are no longer correct.
 
  #if HAS_MARLINUI_U8GLIB
    #ifndef BOARD_ST7920_DELAY_1
      #define BOARD_ST7920_DELAY_1 DELAY_NS(96)
    #endif
    #ifndef BOARD_ST7920_DELAY_2
      #define BOARD_ST7920_DELAY_2 DELAY_NS(48)
    #endif
    #ifndef BOARD_ST7920_DELAY_3
      #define BOARD_ST7920_DELAY_3 DELAY_NS(600)
    #endif
  #endif

comparing it to another stm32f4 I see

  #ifndef BOARD_ST7920_DELAY_1
    #define BOARD_ST7920_DELAY_1    DELAY_NS(125)
  #endif
  #ifndef BOARD_ST7920_DELAY_2
    #define BOARD_ST7920_DELAY_2    DELAY_NS(90)
  #endif
  #ifndef BOARD_ST7920_DELAY_3
    #define BOARD_ST7920_DELAY_3   DELAY_NS(600)
  #endif

try these.

hartmannathan, 2021-05-03:
> #ifndef BOARD_ST7920_DELAY_1
> #define BOARD_ST7920_DELAY_1 DELAY_NS(125)
> #endif
> #ifndef BOARD_ST7920_DELAY_2
> #define BOARD_ST7920_DELAY_2 DELAY_NS(90)
> #endif
> #ifndef BOARD_ST7920_DELAY_3
> #define BOARD_ST7920_DELAY_3 DELAY_NS(600)
> #endif

Thanks for your reply. I will try to test today and if this fixes the issue I'll send a PR.

hartmannathan, 2021-05-04:
With the values above, the display is much improved but still slightly garbled; in other words, It seems the values above are close but not quite correct.

What is the meaning of the three delay values?

thisiskeithb, 2021-05-04:
Maybe we can setup some better defaults. I'm noticing this issue across most boards with a 12864 LCD.

hartmannathan, 2021-05-04:
> Maybe we can setup some better defaults. I'm noticing this issue across most boards with a 12864 LCD.

Is 77966135e81a3f6667de63a04527dc27d1f7fc9a the commit that changed the timings? I am trying to figure out what the old timings were and translate them to the new timings, but I might be going down the wrong rabbit hole.

XDA-Bam, 2021-05-04:
> 
> 
> > Maybe we can setup some better defaults. I'm noticing this issue across most boards with a 12864 LCD.
> 
> Is [7796613](https://github.com/MarlinFirmware/Marlin/commit/77966135e81a3f6667de63a04527dc27d1f7fc9a) the commit that changed the timings? I am trying to figure out what the old timings were and translate them to the new timings, but I might be going down the wrong rabbit hole.

Yes, but you also have to take #21546 into account, which made some delays a bit longer again (#20901 should have reduced shorter delays). Another user tested the delays on his SKR pro and came up with [this](https://github.com/MarlinFirmware/Marlin/issues/21264#issuecomment-795279113), which is close to what ellensp recommended.

I would recommend that you try to first increase delay 1 to 150 ns, then the others one by one by 10 to 20 ns each. Oddly enough, AVR boards run fine with totally different values (125/0/125 instead of 125/90/600). You might also try those first.

X-Ryl669, 2021-05-12:
I agree here. If you read each device datasheet, the expected delay are not like those written in the code. The code was initially hand tuned to account for the wasted cycles in calling the functions and many other. Since now the DELAY_NS is a lot more precise, this tuning does not match anymore with previous code. So you'll need to adjust the timing (we might have missed some). 

If I were you, I would start with what the datasheet says and bootstrap from there. You might need to reduce the delay by *very* small amount (on some ARM board, the cycle step is only 6ns) to account for overhead.

X-Ryl669, 2021-05-12:
The datasheet list 600ns for delay 3, minimum 80ns for delay 2 (+ 2x the rise & fall time, but it's not expressed in the datasheet), and, and 120ns for delay 1 (+ 2x *unknown* rise and fall time). You can take some margin here.

closed 2021-08-11 by thinkyhead

hartmannathan, 2021-08-18:
@thinkyhead was this issue fixed for all boards? The above fix mentioned by @thisiskeithb is for Octopus 12864 only.

thinkyhead, 2021-08-18:
The same solution applies to all cases where the display is garbled, and generally the defaults should be close to the values from the datasheet as described above. If some pins file is specifying delays that are a bit too short, that file may be adjusted for a future release.

hartmannathan, 2021-08-18:
Thanks. I'll test it next time I have an opportunity.