MarlinFirmware/Marlin #21780: [BUG] (Build fails because python looks for build_flags.py in the wrong directory)

GusMcN, 2021-05-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I’m trying to build the Marlin firmware using Auto Build Marlin (2.1.32) and PlatformIO IDE (2.3.2) under VSCode (1.55.2) in Windows 10.
I have all the build files in D:\MFW\Marlin-bugfix-2.0.x. When I try to build, whether from Auto Build Marlin or directly from PlatformIO, I get the following error.

python: can’t open file ‘D:\Marlin\src\HAL\STM32F1\build_flags.py’: [Errno 2] No such file or directory

It should, of course, be looking for
D:\MFW\Marlin-bugfix-2.0.x\Marlin\src\HAL\STM32F1\build_flags.py

Is this a bug or is there some way to direct python to only look under the build directory?
Any help gratefully received.

### Bug Timeline

I'm a newbie and it's been here as long as I have - about a week.

### Expected behavior

I expected it to compile.

### Actual behavior

The compile failed as described above
[BugConfig.zip](https://github.com/MarlinFirmware/Marlin/files/6412039/BugConfig.zip)
[image]

### Version of Marlin Firmware

N/A I'm not having a problem with the firmware. I'm having a problem with the build system. Hopefully I'll advance to having problems with the FW in the fullness of time.

### Printer model

Creality Ender3 v2

### Electronics

N/A

### Add-ons

N/A

### Your Slicer

Cura

### Host Software

OctoPrint

rhapsodyv, 2021-05-02:
can you test this: https://github.com/rhapsodyv/Marlin/tree/fix-script-path ?

GusMcN, 2021-05-03:
Slightly different. 
I attempted the default mega2560 build and got this variation (see attached JPG)
Different file but problem still seems to emanate from 

File "C:\Users\Gus\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 175:

[image]

rhapsodyv, 2021-05-03:
Your platformio install seems corrupted...
Can you reinstall everything?

GusMcN, 2021-05-03:
Did that, several times now - made no difference.
The problem seems to be that avr-g++.exe doesn't know to look for files under the D:/MFW/Marlin-fix-script-path directory. I see that it has a -B flag to add a directory to its search path. Is there any way to enable this?
Is there an environment variable I need to set/unset?
What is the difference between my setup and yours?

rhapsodyv, 2021-05-03:
Ok, let's recap:

Your PIO install is in `C:\Users\Gus\.platformio\`
Marlin is located at: `D:/MFW/Marlin-fix-script-path`

I will try to simulate it, but I suspect PIO is having a bad time to work when it's installed in a different drive than the project.

GusMcN, 2021-05-05:
I've made some progress.
The basic problem is that when avr-g++.exe runs the current working directory(CWD) is D:\
I proved this by renaming avr-g++.exe to avr-g+++.exe and replacing it with my own version which displays the CWD.
In my version I set the CWD to D:/MFW/Marlin-fix-script-path and then executed the renamed version. This solved most of my problems but it looks like ar.exe may need the same treatment. See Output.jpg.
While I can probably rig up a system that will let me compile it's hardly a general solution. There must be others in the same position as myself.
So the question is why does the CWD revert to D:\ and is there some way to force it to be correct.

[main.cpp.pdf](https://github.com/MarlinFirmware/Marlin/files/6429685/main.cpp.pdf)
[image]

GusMcN, 2021-05-09:
Mea culpa, mea culpa, mea maxima culpa.
I've worked out what's happening.
When I'm working in a command console I like to have certain Doskey macros enabled and some other bits of environment.
I arrange this by naming a batch file in HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Command Processor\AutoRun.
A while ago I set this batch file to set the directory to D:\ and that's what's been happening.
I'll post this here as a warning to others.

GusMcN, 2021-05-09:
Hi Victor

I’ve posted the resolution to this – all my fault.

Thanks for taking the time  to look at it

 

Cheers

Gus

closed 2021-05-09 by GusMcN