MarlinFirmware/Marlin #21782: [BUG] MKS NANO V3 W/TS35 V2.0  Hangs on firmware installation

Savage79, 2021-05-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Latest Bugfix and Latest 2.0.8 Firmware both same issue, compile no errors, install on SD card or USB stick, power on, goes through firmware installation, Hangs on 100% and does not initiate Marlin.  No action fixes issue, must install older version of Bugfix Downloaded April 22/2021. and that version works.

Also same error happens regardless the UI of the screen. TFT_FULL_COLOR or TFT_LVGL 

### Bug Timeline

New with 2.0.8

### Expected behavior

Finish installation as required

### Actual behavior

[image]
Hangs at this location

### Steps to Reproduce

Compile firmware, NO errors
Copy to SD Card or USB 
Insert drive to board and power on
Board shows 100% and give 2 confirmation tones, and hangs.

### Version of Marlin Firmware

Marlin Bugfix 2.0.X and 2.0.8 Latest  (May 2,2021) 

### Printer model

Flsun QQS Pro

### Electronics

MKS Nano V3 w/TS35 V2 and TMC2209 Uart

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

[Marlin208.zip](https://github.com/MarlinFirmware/Marlin/files/6412189/Marlin208.zip)
[Marlinbugfix208.zip](https://github.com/MarlinFirmware/Marlin/files/6412190/Marlinbugfix208.zip)

rhapsodyv, 2021-05-02:
I think I fixed it today with #21775 and #21776

Savage79, 2021-05-02:
I will give it a go here shortly.

Savage79, 2021-05-03:
@rhapsodyv the fix that you applied does not fix this hang issue that I have.

rhapsodyv, 2021-05-03:
I will test with your configs. But for me, it's working fine.

Savage79, 2021-05-04:
@rhapsodyv I downloaded a fresh copy tonight, and compiled, and perfect, the board took the new firmware.

Thank you and everyone for taking a look at this issue.  Your work doesn't go un-noticed.

rhapsodyv, 2021-05-04:
So this can be close? :-)

Savage79, 2021-05-04:
I would assume so :)

closed 2021-05-04 by rhapsodyv