MarlinFirmware/Marlin #21783: [BUG]Marlin 2 ‘bug ：X axis and axis layer offset

chinamore, 2021-05-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin 2 ‘bug ：X axis and axis layer offset

### Actual behavior

Marlin 2 ‘bug ：X axis and axis layer offset

### Steps to Reproduce

Motherboard: MKS GEN L 2.1
MARLIN: bugfix-2.0.x
Scenario: When the motherboard uses an independent stepper driver and a hybrid servo driver (HBS57)

### Version of Marlin Firmware

2.0

### Your Slicer

Slic3r

### Host Software

Repetier Host

### Additional information & file uploads

Marlin 2 ‘bug ：X axis and axis layer offset

chinamore, 2021-05-03:
https://www.m175.kr/blog/zb_users/upload/2021/05/202105031620005230490261.jpg

chinamore, 2021-05-03:
It took me 3 months to prove that it has nothing to do with my hardware. I redo all the mechanical parts and reduce

chinamore, 2021-05-03:
Machine picture：
https://www.m175.kr/blog/zb_users/upload/2021/05/202105031620005901488963.jpg
Print picture
https://www.m175.kr/blog/zb_users/upload/2021/05/202105031620005901774180.jpg

When you print a super large graphic, the offset is also particularly obvious. Cube (200MM*200MM*200MM, if the graphic is too small, it is not obvious than a 12mm cylinder, and the print is still straight.

dwareb, 2021-05-06:
Have you ruled out issues related to the stepper motors? Like, say you swap which one controls x, and which controls y, does the problem follow it? If you have a spare controller board for the printer, might try that too.

closed 2021-07-23 by github-actions[bot]