MarlinFirmware/Marlin #21786: [BUG] (bug summary) no media

Robenc, 2021-05-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

getting no media

### Bug Timeline

Stasrted with 2.0.8

### Actual behavior

its not reading the SD card on the on board slot

### Version of Marlin Firmware

2.0.8

### Printer model

Creality Ender 3

### Electronics

BTT Skr Mini e3 v2

ellensp, 2021-05-03:
Did you set #define SDCARD_CONNECTION ONBOARD  and  #define SUPPORT ?

ellensp, 2021-05-03:
Please attach your configuration files.

Robenc, 2021-05-03:
files are the ones in the config not modified
[BigTreeTech SKR Mini E3 2.0.zip](https://github.com/MarlinFirmware/Marlin/files/6415361/BigTreeTech.SKR.Mini.E3.2.0.zip)

Robenc, 2021-05-03:
> 
> 
> Please attach your configuration files.

just uploaded

Robenc, 2021-05-03:
> 
> 
> Did you set #define SDCARD_CONNECTION ONBOARD and #define SDSUPPORT ?

could it be that the configuration files online are missing some coding ?

Robenc, 2021-05-07:
connected printer to pc and issued a m21 and it sees the card fine shutdown print power on an no media again
[image]

Robenc, 2021-05-07:
[image]

Robenc, 2021-05-07:
after m21

[image]

Robenc, 2021-05-07:
a little more help on the issue at hand

Robenc, 2021-05-07:
> 
> 
> Please attach your configuration files.

sent more info to help

rhapsodyv, 2021-05-08:
I have a SKR MINI E3 V2. I just tested your configs and everything worked fine with current bugfix. 

I think you have a hardware issue.

closed 2021-05-08 by rhapsodyv