MarlinFirmware/Marlin #21787: [BUG] Warning! More than one package has been found by toolchain-gccarmnoneeabi @ >=1.90000 requirements:

Marlor, 2021-05-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

A fresh (no cache in platformio) Build of 2.0.8 and bugfix-2.0.x for LPC1768 gives the following warning:

```
Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1768; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing toolchain-gccarmnoneeabi @ >=1.90000
Tool Manager: Warning! More than one package has been found by toolchain-gccarmnoneeabi @ >=1.90000 requirements:        
 - platformio/toolchain-gccarmnoneeabi @ 1.50401.210106
 - maxgerhardt/toolchain-gccarmnoneeabi @ 1.40903.0
Tool Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-gccarmnoneeabi @ 1.90301.200702 has been installed!
```

The only changes i made are
`#define MOTHERBOARD BOARD_BTT_SKR_V1_3`
in Configuration.h
and 
`default_envs = LPC1768`
in platformio.ini

It's just a warning but i wanted to let you know.
Most people will probably newer see it because of the cache in platformio.

### Version of Marlin Firmware

bugfix-2.0.x

rhapsodyv, 2021-05-08:
This is not a Marlin issue. It's a https://github.com/p3p/pio-nxplpc-arduino-lpc176x issue. @p3p, can you fix this?

Marlor, 2021-07-10:
Same as before

ellensp, 2021-07-10:
This is not a marlin issue as already pointed out, as such I am closing this. But I have created issue in correct github repository
https://github.com/p3p/pio-nxplpc-arduino-lpc176x/issues/8

closed 2021-07-10 by ellensp