MarlinFirmware/Marlin #21788: [BUG] Assisted tramming validation bug v2.0.8

sammy2142, 2021-05-03:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

The current checks for bed limits are incorrect. The check for the left side of the bed is correct but the right side is incorrect for the x-axis.
My probe has an x offset of -42.0 (so the probe is to the left side of the nozzle).
If you set PROBING_MARGIN 0 and just use the inbuilt checks the probe points on the left side of the bed are correctly limited to nozzle position of x >= 42 so the probe doesn't run off the bed.
What is incorrect is this same limit is applied on the right side of the bed. Since the probe is on the left of the nozzle even if you set x to the bed max (235 in my case) the probe will never run off the bed. 
So the max limit for the probe on the right side should be the nozzle bed limit and not (nozzle bed limit - x offset).
Please see attached picture below.
Y limits should also be looked at.

### Bug Timeline

v2.0.8

### Version of Marlin Firmware

v2.0.8

### Printer model

Ender 3

### Electronics

BTT mini skr e3 v2.0

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[image]

Marlor, 2021-05-03:
Can not confirm this Bug.

Have a look if you have set the right Travel limits in addition to bed size.
This way the Probe has a wider range of motion

thisiskeithb, 2021-05-03:
Please attach your configs like the bug report template asked.

thisiskeithb, 2021-05-03:
Looking at this more closely, this isn't a bug. If your hardware allows & you want to probe more of the area on the right side of the bed, increase `X_MAX_POS`.

closed 2021-05-03 by thisiskeithb

sammy2142, 2021-05-04:
Hi @thisiskeithb & @Marlor 
Could you have another look at this please?
My config files:
[config_marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6420295/config_marlin.zip)

My X_MAX_POS is correctly set
#define X_BED_SIZE 235 
#define X_MAX_POS X_BED_SIZE

The build error is:
`In file included from Marlin\src\feature\tramming.cpp:27:0:
Marlin\src\feature\tramming.h:36:36: error: static assertion failed: TRAMMING_POINT_XY point 1 is not reachable with the default NOZZLE_TO_PROBE offset and PROBING_MARGIN.
 #define VALIDATE_TRAMMING_POINT(N) static_assert(N >= G35_PROBE_COUNT || Probe::build_time::can_reach(screws_tilt_adjust_pos[N]), \
                                    ^
Marlin\src\feature\tramming.h:38:29: note: in expansion of macro 'VALIDATE_TRAMMING_POINT'
 VALIDATE_TRAMMING_POINT(0); VALIDATE_TRAMMING_POINT(1); VALIDATE_TRAMMING_POINT(2); VALIDATE_TRAMMING_POINT(3); VALIDATE_TRAMMING_POINT(4); VALIDATE_TRAMMING_POINT(5);
                             ^~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\feature\tramming.h:36:36: error: static assertion failed: TRAMMING_POINT_XY point 2 is not reachable with the default NOZZLE_TO_PROBE offset and PROBING_MARGIN.
 #define VALIDATE_TRAMMING_POINT(N) static_assert(N >= G35_PROBE_COUNT || Probe::build_time::can_reach(screws_tilt_adjust_pos[N]), \`

Doing some more testing it definitely seems the validation check for tramming points is setting nozzle movement limits on all 4 sides of the bed.

If my probe is to the left and in front of the nozzle, the probe can only go off the bed at the left and front side of the bed in normal nozzle xy range.
Another example is if your probe is to the right and behind the nozzle, the probe can only go off the bed at the right and back side of the bed in normal nozzle xy range.

For the validation check you only need to apply nozzle movement limits on at most 2 sides of the bed (not all sides).

Ps. my PROBING_MARGIN is set to 0 so shouldn't affect the validation

Marlor, 2021-05-04:
`X_MAX_POS` is not your `X_BED_SIZE`

Use manual move to see how far your gantry really can go and set `X_MAX_POS` accordingly
Remember to disable soft endstops

sammy2142, 2021-05-04:
Firmware is not building because of the tramming validation error. So cant calibrate x max and min etc yet.

Marlor, 2021-05-04:
Disable tramming in the meantime, you need to reflash anyway after you found the right `X_MAX_POS/Y_MAX_POS`

sammy2142, 2021-05-04:
My gantry is not massively out anyway maybe max 1 step - so I'm not too worried about calibrating this. It shouldn't cause the tramming validation error on build though.

Marlor, 2021-05-04:
Your TRAMMING_POINT_XY x point is on 235
Your PROBE_OFFSET for x is -42
Your X_MAX_POS is 235

Translated: Probe 42 mm further out than the Travel limits allow

sammy2142, 2021-05-04:
> Your TRAMMING_POINT_XY x point is on 235
> Your PROBE_OFFSET for x is -42
> Your X_MAX_POS is 235
> 
> Translated: Probe 42 mm further out than the Travel limits allow

No
The probe is to the left of the nozzle - the nozzle can go right to 235 and the probe is still on the bed :)
This is the check that's the problem ;)

sammy2142, 2021-05-04:
x min should be limited to 42 as this the side of the bed where the nozzle can go off
235 is fine.
Regards

Marlor, 2021-05-04:
Thats not what you wrote in your Config
Can the Probe phisicaly hit 235?

sammy2142, 2021-05-04:
The nozzle can hit 235 and the probe will be at 193 at this position since it is 42 to the left regards - it may be better if I take a picture if this is not clear.

Marlor, 2021-05-04:
TRAMMING_POINT_XY are coordiantes to Probe
You are telling Marlin to probe 235 which is not possible

sammy2142, 2021-05-04:
Ok thanks I will check this - I thought they were nozzle positions (and I was calculating where the probe would be).
I will report back if this fixes it.

sammy2142, 2021-05-04:
Ok my bad it woks when setting probe point locations and not nozzle points. 
I lost my old config file and got my tramming points back from running g35 and writing down the locations (and of course these were the nozzle locations).
Thanks for helping me
Regards