MarlinFirmware/Marlin #21789: [BUG] CASE_LIGHT_USE_RGB_LED error

sanek88lbl, 2021-05-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin don't build for Lerdge K with **_CASE_LIGHT_USE_RGB_LED_**.

### Bug Timeline

Begins from

### Expected behavior

```
Environment    Status    Duration
-------------  --------  ------------
LERDGEK        SUCCESS   00:00:55.274
```

### Actual behavior

```
Executing task: C:\Users\user\.platformio\penv\Scripts\platformio.exe run --environment LERDGEK <

Processing LERDGEK (platform: ststm32@~12.0; board: marlin_STM32F407ZGT6; framework: arduino)
------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_STM32F407ZGT6.html
PLATFORM: ST STM32 (12.0.0) > STM32F407ZGT6(192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200819 (1.9.0)
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TMCStepper> 0.7.1
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
|-- <EEPROM> 2.0.1
Building in release mode
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\HAL.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\HAL_MinSerial.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\HAL_SPI.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\MarlinSPI.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\MarlinSerial.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\Sd2Card_sdio_stm32duino.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\Servo.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\eeprom_flash.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\eeprom_sdcard.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\eeprom_sram.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\eeprom_wired.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\fast_pwm.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\fastio.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\msc_sd.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\tft\tft_fsmc.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\tft\tft_ltdc.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\tft\tft_spi.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\tft\xpt2046.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\timers.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\usb_host.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\usb_serial.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\watchdog.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\shared\HAL_MinSerial.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\LERDGEK\src\src\MarlinCore.cpp.o
Compiling .pio\build\LERDGEK\src\src\core\serial.cpp.o
Compiling .pio\build\LERDGEK\src\src\core\utility.cpp.o
Compiling .pio\build\LERDGEK\src\src\feature\bedlevel\abl\abl.cpp.o
Compiling .pio\build\LERDGEK\src\src\feature\bedlevel\bedlevel.cpp.o
Compiling .pio\build\LERDGEK\src\src\feature\bltouch.cpp.o
Compiling .pio\build\LERDGEK\src\src\feature\caselight.cpp.o
Compiling .pio\build\LERDGEK\src\src\feature\controllerfan.cpp.o
In file included from Marlin\src\feature\../inc/MarlinConfigPre.h:56,
                 from Marlin\src\feature\../inc/MarlinConfig.h:28,
                 from Marlin\src\feature\caselight.cpp:23:
Marlin\src\feature\../inc/../../Configuration_adv.h:592:59: error: could not convert '{255, 255, 255, 255}' from '<brace-enclosed initializer list>' to 'LEDColor'
  592 |     #define CASE_LIGHT_DEFAULT_COLOR { 255, 255, 255, 255 } // { Red, Green, Blue, White }
      |                                                           ^
      |                                                           |
      |                                                           <brace-enclosed initializer list>
Marlin\src\feature\caselight.cpp:48:7: note: in expansion of macro 'CASE_LIGHT_DEFAULT_COLOR'
   48 |       CASE_LIGHT_DEFAULT_COLOR
      |       ^~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\feature\caselight.cpp: In static member function 'static void CaseLight::update(bool)':
Marlin\src\feature\caselight.cpp:83:5: error: no matching function for call to 'LEDLights::set_color(LEDColor, bool)'   
   83 |     );
      |     ^
In file included from Marlin\src\feature\caselight.h:27,
                 from Marlin\src\feature\caselight.cpp:27:
Marlin\src\feature\leds/leds.h:149:15: note: candidate: 'static void LEDLights::set_color(const LEDColor&)'
  149 |   static void set_color(const LEDColor &color
      |               ^~~~~~~~~
Marlin\src\feature\leds/leds.h:149:15: note:   candidate expects 1 argument, 2 provided
Marlin\src\feature\leds/leds.h:155:22: note: candidate: 'static void LEDLights::set_color(uint8_t, uint8_t, uint8_t)'   
  155 |   static inline void set_color(uint8_t r, uint8_t g, uint8_t b
      |                      ^~~~~~~~~
Marlin\src\feature\leds/leds.h:155:22: note:   candidate expects 3 arguments, 2 provided
Marlin\src\feature\caselight.cpp:75:44: warning: unused variable 'n10ct' [-Wunused-variable]
   75 |     const uint8_t i = on ? brightness : 0, n10ct = INVERT_CASE_LIGHT ? 255 - i : i;
      |                                            ^~~~~
*** [.pio\build\LERDGEK\src\src\feature\caselight.cpp.o] Error 1
============================================= [FAILED] Took 15.96 seconds =============================================

Environment    Status    Duration
-------------  --------  ------------
LERDGEK        FAILED    00:00:15.963
======================================== 1 failed, 0 succeeded in 00:00:15.963 ======================================== 
```

### Steps to Reproduce

1.  Clone [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip).
2.  Replace _Configuration.h_, _Configuration_adv.h_ and _pins_LERDGE_K.h_ from my [Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6416766/Marlin.zip).

3. Start **_Build_** from **_PROJECT TASKS -> LERDGEK->General_**.

### Version of Marlin Firmware

M115 Recv: FIRMWARE_NAME:Marlin 2.0.8 (May  3 2021 20:27:31) 

### Printer model

Custom, B&R

### Electronics

Lerdge K board

### Host Software

OctoPrint

### Additional information & file uploads

You need to try to make a build for other boards with RGB LED.

shitcreek, 2021-05-05:
Number of argument and function overloading conflicts. And it looks like the case light stuff utilizes the same led stuff but there isn't a condition to enable the led if case light is enabled.

I got it to compile without errors with these changes, but have not verified if it works or conflicts with anything else so test at your own risk:

Change line 56 of `leds.cpp` and  line 190 of `leds.h` from:
`#if EITHER(LED_CONTROL_MENU, PRINTER_EVENT_LEDS)`
to:
`#if ANY(LED_CONTROL_MENU, PRINTER_EVENT_LEDS, CASE_LIGHT_ENABLE)`

Change line 192 of `Conditionals_adv.h` from:
`#if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)`
to:
`#if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED, CASE_LIGHT_ENABLE)`

Change line 50 of `caselight.cpp` from:
` { 255, 255, 255, 255 }`
to:
```
      { 255, 255, 255
      #if HAS_WHITE_LED
      , 255
      #endif
      }
```

sanek88lbl, 2021-05-05:
> Number of argument and function overloading conflicts. And it looks like the case light stuff utilizes the same led stuff but there isn't a condition to enable the led if case light is enabled.
>
> I [...]

Sorry, but it didn't help:

```
In file included from Marlin\src\feature\../inc/MarlinConfigPre.h:56,
                 from Marlin\src\feature\../inc/MarlinConfig.h:28,
                 from Marlin\src\feature\caselight.cpp:23:
Marlin\src\feature\../inc/../../Configuration_adv.h:592:59: error: could not convert '{255, 255, 255, 255}' from '<brace-enclosed initializer list>' to 'LEDColor'
  592 |     #define CASE_LIGHT_DEFAULT_COLOR { 255, 255, 255, 255 } // { Red, Green, Blue, White }
      |                                                           ^
      |                                                           |
      |                                                           <brace-enclosed initializer list>
Marlin\src\feature\caselight.cpp:48:7: note: in expansion of macro 'CASE_LIGHT_DEFAULT_COLOR'
   48 |       CASE_LIGHT_DEFAULT_COLOR
      |       ^~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\feature\caselight.cpp: In static member function 'static void CaseLight::update(bool)':
Marlin\src\feature\caselight.cpp:87:5: error: no matching function for call to 'LEDLights::set_color(LEDColor, bool)'   
   87 |     );
      |     ^
In file included from Marlin\src\feature\caselight.h:27,
                 from Marlin\src\feature\caselight.cpp:27:
Marlin\src\feature\leds/leds.h:149:15: note: candidate: 'static void LEDLights::set_color(const LEDColor&)'
  149 |   static void set_color(const LEDColor &color
      |               ^~~~~~~~~
Marlin\src\feature\leds/leds.h:149:15: note:   candidate expects 1 argument, 2 provided
Marlin\src\feature\leds/leds.h:155:22: note: candidate: 'static void LEDLights::set_color(uint8_t, uint8_t, uint8_t)'   
  155 |   static inline void set_color(uint8_t r, uint8_t g, uint8_t b
      |                      ^~~~~~~~~
Marlin\src\feature\leds/leds.h:155:22: note:   candidate expects 3 arguments, 2 provided
Marlin\src\feature\caselight.cpp:79:44: warning: unused variable 'n10ct' [-Wunused-variable]
   79 |     const uint8_t i = on ? brightness : 0, n10ct = INVERT_CASE_LIGHT ? 255 - i : i;
      |                                            ^~~~~
*** [.pio\build\LERDGEK\src\src\feature\caselight.cpp.o] Error 1
```

```
Marlin\src\feature\../inc/../../Configuration_adv.h:592:59: error: could not convert '{255, 255, 255, 255}' from '<brace-enclosed initializer list>' to 'LEDColor'
  592 |     #define CASE_LIGHT_DEFAULT_COLOR { 255, 255, 255, 255 } // { Red, Green, Blue, White }
```

By the way, this error can be corrected by replacing line 592 in _Configuration_adv.h_ on this one:
`    #define CASE_LIGHT_DEFAULT_COLOR {{ 255, 255, 255, 255 }} // { Red, Green, Blue, White }`

I will attach my full project to this message:
[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/6425179/Marlin-bugfix-2.0.x.zip)

ellensp, 2021-05-05:
Comment out #define CASE_LIGHT_DEFAULT_COLOR in configuration_adv.h you don't need it unless you want other colours.
In that case drop the last digit so it becomes #define CASE_LIGHT_DEFAULT_COLOR { 255, 255, 255 }  as your LCD's don't have a physical white LED

sanek88lbl, 2021-05-05:
> Comment out #define CASE_LIGHT_DEFAULT_COLOR in configuration_adv.h you don't need it unless you want other colour as default.
> If you do want a different default drop the last digit so it becomes #define CASE_LIGHT_DEFAULT_COLOR { 255, 255, 255 } as your LCD's don't have a physical white LED

Thank you, it didn't help.

sanek88lbl, 2021-05-05:
> Number of argument and function overloading conflicts. And it looks like the case light stuff utilizes the same led stuff but there isn't a condition to enable the led if case light is enabled.
> ....

I fixed this bug:
1) changed from line 80 of _caselight.cpp_:
from:
```
    leds.set_color(
      MakeLEDColor(color.r, color.g, color.b, color.w, n10ct),
      false
    );
```
to:
```
    leds.set_color(
      MakeLEDColor(color.r, color.g, color.b, color.w, n10ct)
      #if ENABLED(NEOPIXEL_LED)
        , false
      #endif
    );
```
2) changed line 582 of _Configuration_adv.h_:
from:
```
  //#define CASE_LIGHT_NO_BRIGHTNESS          // Disable brightness control. Enable for non-PWM lighting.
```
to:
```
  #define CASE_LIGHT_NO_BRIGHTNESS          // Disable brightness control. Enable for non-PWM lighting.
```
3) used your tips (they don't solve the problem, but they look necessary)

sanek88lbl, 2021-05-05:
If anyone knows how to add this to the branch, help me)

I'm using VS Code with GIT on Win10. I cloned the _bugfix-2.0.x_ branch and am fixing this bug only  locally on my machine.

ellensp, 2021-05-05:
go back to bugfix + your configs and pins

Edit Configuration_adv.h

change     #define CASE_LIGHT_DEFAULT_COLOR { 255, 255, 255, 255 } // { Red, Green, Blue, White }
to   #define CASE_LIGHT_DEFAULT_COLOR { 255, 255, 255} // { Red, Green, Blue}
change   //#define CASE_LIGHT_NO_BRIGHTNESS          // Disable brightness control. Enable for non-PWM lighting.
to  #define CASE_LIGHT_NO_BRIGHTNESS          // Disable brightness control. Enable for non-PWM lighting.

change Marlin/src/feature/caselight.cpp
```
    leds.set_color(
      MakeLEDColor(color.r, color.g, color.b, color.w, n10ct),
      false
    );
```
to
```
    leds.set_color(
      MakeLEDColor(color.r, color.g, color.b, color.w, n10ct)
    );

```

closed 2021-05-05 by sanek88lbl