MarlinFirmware/Marlin #21790: [BUG] TFT32 print doesn't work, Homeing etc. works with TFT

cprobst87, 2021-05-03:
Hello,
I installed Marlin 2.0.8
Now my TFT32 doesn't work correctly: I can do Homeing, Heating etc. But if I start a print, it hangs from the beginning. No reaction. I can cancel the print, it will home. 
I tried other baudrates, the same.
Is there for example a new status line sending to the serial port?
With the release before, it works fine.
Config files attached.
Thank you!
[Config.zip](https://github.com/MarlinFirmware/Configurations/files/6416546/Config.zip)

rhapsodyv, 2021-05-03:
I can confirm this behaviour. But it's not clear that it's a Marlin issue or not. If the TFT firmware better handle `resend` requests, everything works.

Anyway, it seems that something in recent marlin changed that is triggering this issue with serial TFTs. I'm talking with @makerbase-mks to add a fix to their TFT firmware, that is fast and easy. But it probably will affect all serial TFTs out there... So we need to investigate it in the Marlin side.

Options:
 1) Older marlin silent lose data and now it complain (that is good)

 2) Newer Marlin is losing a lot more data than before to the point that it’s generating a lot of resend requests, breaking serial tfts (that is bad)

3) Newer marlin is generating false resend requests breaking (that is bad)

4) Newer marlin changed its behavior of resend requests (that is bad too, because breaks years of compatibility)

cprobst87, 2021-05-03:
With Marlin 1.1.9 there was a similar bug: https://github.com/makerbase-mks/MKS-TFT/issues/190#issuecomment-425693213

Quote from 2018:
"TFT32 sends: N-1 M110*15
Marlin Replies: ok N-1 P15 B3

(After 15 seconds)
TFT32 sends: N-1 M110*15
Marlin Replies: ok N-1 P15 B3

(After another 15 seconds)
TFT32 sends: N-1 M110*15
Marlin Replies: ok N-1 P15 B3

At this point, I stopped the capture."_

Maybe this can help.

rhapsodyv, 2021-05-03:
Can you test marlin 2.0.7.2 or even 2.0.6.2?

cprobst87, 2021-05-04:
Hi, Marlin 2.0.7.2 works fine.

DenDeg, 2021-05-04:
I have same issue with Makerbase MKS-TFT3.5 and Marlin 2.0.8. But in my case Marlin 2.0.7.2 doesn't work too.  Marlin 2.0.6.1 works fine.

ellensp, 2021-05-04:
If this is resend related, there was a missing OK that was patched 2 hours ago. https://github.com/MarlinFirmware/Marlin/commit/53fc13ba7fa46f108be19ccde206dd46741e07be
re download Marlin 2.0.x and try again   https://github.com/MarlinFirmware/Marlin/archive/refs/heads/2.0.x.zip

cprobst87, 2021-05-04:
That was the problem, now it prints.
But new error: Checksum Error on Line Number 6... I can confirm that and it'll print.
I'll test it again.

cprobst87, 2021-05-04:
It works fine! The missing "OK" was the solution for this bug. 
Thank you all!

closed 2021-05-04 by cprobst87