MarlinFirmware/Marlin #21795: [BUG] Cant compile MMU2 cause idle was not declared

Lyr3x, 2021-05-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I yesterday synced to the latest bugfix-2.0.x branch (old one is about 5 month old). Triple checked the configuration changes but I can't compile marlin when uncommenting `#define MMU_MODEL PRUSA_MMU2`

Error:
```
Marlin/src/lcd/menu/menu_mmu2.cpp: In function 'void action_mmu2_unload_filament()':
Marlin/src/lcd/menu/menu_mmu2.cpp:85:3: error: 'idle' was not declared in this scope
   85 |   idle();
      |   ^~~~
Marlin/src/lcd/menu/menu_mmu2.cpp: In function 'void mmu2_M600()':
Marlin/src/lcd/menu/menu_mmu2.cpp:158:29: error: 'idle' was not declared in this scope
  158 |   while (wait_for_mmu_menu) idle();
      |                             ^~~~
Marlin/src/lcd/menu/menu_mmu2.cpp: In function 'uint8_t mmu2_choose_filament()':
Marlin/src/lcd/menu/menu_mmu2.cpp:165:29: error: 'idle' was not declared in this scope
  165 |   while (wait_for_mmu_menu) idle();
      |                             ^~~~
*** [.pio/build/BIGTREE_SKR_PRO/src/src/lcd/menu/menu_mmu2.cpp.o] Error 1
```

`#define NOZZLE_PARK_FEATURE` is activated so are the advanced options like `#define MMU2_MENUS`

### Bug Timeline

Seems to be a fairly new bug

### Expected behavior

Compiling succeed without errors

### Actual behavior

Compile stopped with errors

### Version of Marlin Firmware

53fc13ba7fa46f108be19ccde206dd46741e07be

### Printer model

HevoRS

### Electronics

SKR Pro 1.1

### Additional information & file uploads

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6419726/configuration.zip)

Complete config: https://github.com/Lyr3x/Marlin/blob/bugfix-2.0.x-mmu/

Minimal changes: https://github.com/Lyr3x/Marlin/blob/bugfix-2.0.x-mmu-debug/

ellensp, 2021-05-04:
Give this a try, in  Marlin/src/lcd/menu/menu_mmu2.cpp  add the line #include "../../MarlinCore.h"

ie
```CPP
diff --git a/Marlin/src/lcd/menu/menu_mmu2.cpp b/Marlin/src/lcd/menu/menu_mmu2.cpp
index 7e71f00d25..af3d9232b2 100644
--- a/Marlin/src/lcd/menu/menu_mmu2.cpp
+++ b/Marlin/src/lcd/menu/menu_mmu2.cpp
@@ -24,6 +24,7 @@
 
 #if BOTH(HAS_LCD_MENU, MMU2_MENUS)
 
+#include "../../MarlinCore.h"
 #include "../../feature/mmu/mmu2.h"
 #include "menu_mmu2.h"
 #include "menu_item.h"

```

Lyr3x, 2021-05-04:
That was an easy one. Ill test that later today and update. Ty for the hint!

Lyr3x, 2021-05-04:
Didnt print anything but communication with the MMU is working fine. Tested loading and unloading of filament. Should i provide a PR?

ellensp, 2021-05-04:
Go for it.

ellensp, 2021-05-05:
Since this still hasn't been created I have done it.

ellensp, 2021-05-06:
Merged, closing issue

closed 2021-05-06 by ellensp