MarlinFirmware/Marlin #21799: Problems with bed temperature when print[BUG] (bug summary)

Ramon-Balaguer, 2021-05-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I manually set the temperature of the bed, the bed perfectly maintains the temperature for minutes (I have reached 30 minutes without problem).
Exmaple this image:
[image]

But when I print (only at high temperatures with the bed at 60 it has not happened to me) the bed after a minute reduces the temperature.
[image]

So I have tried uploading manually, but it is not responding

### Bug Timeline

New bug

### Expected behavior

Maintain printing temperature

### Actual behavior

A few minutes after starting printing, the temperature is reduced.

### Steps to Reproduce

1. Start print
2. Past 3 minuts
3. Temperature of my bed decrease

### Version of Marlin Firmware

Marlin 2.0.8 (May  4 2021 17:43:46)

### Printer model

Geeetech Prusa I3 pro W

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6422573/Marlin.zip)

I have disabled THERMAL PROTECTION BED to be able to print

Ramon-Balaguer, 2021-05-04:
Another example with a larger print.
[image]

Ramon-Balaguer, 2021-05-04:
Terminal with error temperature:
[image]

Ramon-Balaguer, 2021-05-04:
1h printing:
[image]

thisiskeithb, 2021-05-04:
Is this a new bug? Being able to maintain bed temp when nothing else is being powered is kind of a red herring since you're only drawing a fraction of normal power.

If your hardware is normally able to reach 150°C on the bed as you have it configured, have you tried enabling `PIDTEMPBED`, running a PID tune on the bed at the desired printing temp (`M303 E-1 C8 S120`), & then testing again?

Ramon-Balaguer, 2021-05-04:
Hello @thisiskeithb 

I've tried a couple of times what you told me, the printer crashes and starts beeping

[image]

thisiskeithb, 2021-05-04:
Is this a new bug? If so, when did the issue occur?

Is your hardware is normally able to reach 150°C on the bed as you have it configured?

Ramon-Balaguer, 2021-05-05:
With the original firmware it did not happen.

Even so, I have bought a more powerful power supplier and today I will replace it.

Ramon-Balaguer, 2021-05-06:
Hello @thisiskeithb i change PowerSupplier (180W to 360W)and continue with the same problem, when i set manually 240grades for hotend and 100 for bed, i don't have problem, the temperature is maintained.

Ramon-Balaguer, 2021-05-06:
[image]

Ramon-Balaguer, 2021-05-06:
New i try M303 and fail with beeps
[image]

Ramon-Balaguer, 2021-05-06:
What can I do to give you more information?

Ramon-Balaguer, 2021-05-06:
Printing now: 
[image]

malcinator, 2021-05-08:
Hey, I'd also like to jump in on this. I'm experiencing a similar issue. I'm trying to PID tune my two printers and it fails. I have two identical machines (both were Prusa i3 MK2.5s and have been upgraded to SKR1.4Turbo with Bear Upgrades and cloned BMG) and I see the same error across both. The Bed Thermistors measures 75ohms & the beds are measured at 1ohm. I am at this point at a complete loss as what else I can investigate. I have tried the PID first out in an open room at 24.5ºc and then again in side an enclosure where the temperature was around 33ºC.

Im at a complete loss with this and have tried various combinations of the Thermal Settings in Config.adv.h and all fail.

```
2021-05-08 12:43:16,608 - Recv:  T:29.13 /0.00 B:65.00 /0.00 @:0 B@:0
2021-05-08 12:43:18,608 - Recv:  T:29.25 /0.00 B:64.86 /0.00 @:0 B@:127
2021-05-08 12:43:20,608 - Recv:  T:29.13 /0.00 B:64.68 /0.00 @:0 B@:127
2021-05-08 12:43:22,608 - Recv:  T:29.13 /0.00 B:64.54 /0.00 @:0 B@:127
2021-05-08 12:43:24,608 - Recv:  T:29.25 /0.00 B:64.49 /0.00 @:0 B@:127
2021-05-08 12:43:26,608 - Recv:  T:29.38 /0.00 B:64.44 /0.00 @:0 B@:127
2021-05-08 12:43:28,608 - Recv:  T:29.38 /0.00 B:64.54 /0.00 @:0 B@:127
2021-05-08 12:43:30,608 - Recv:  T:29.25 /0.00 B:64.68 /0.00 @:0 B@:127
2021-05-08 12:43:32,608 - Recv:  T:29.25 /0.00 B:64.86 /0.00 @:0 B@:127
2021-05-08 12:43:34,609 - Recv:  T:29.38 /0.00 B:65.04 /0.00 @:0 B@:0
2021-05-08 12:43:36,609 - Recv:  T:29.25 /0.00 B:65.30 /0.00 @:0 B@:0
2021-05-08 12:43:38,609 - Recv:  T:29.38 /0.00 B:65.47 /0.00 @:0 B@:0
2021-05-08 12:43:40,609 - Recv:  T:29.50 /0.00 B:65.69 /0.00 @:0 B@:0
2021-05-08 12:43:42,609 - Recv:  T:29.38 /0.00 B:65.86 /0.00 @:0 B@:0
2021-05-08 12:43:44,609 - Recv:  T:29.50 /0.00 B:65.95 /0.00 @:0 B@:0
2021-05-08 12:43:46,609 - Recv:  T:29.50 /0.00 B:65.95 /0.00 @:0 B@:0
2021-05-08 12:43:48,609 - Recv:  T:29.50 /0.00 B:65.95 /0.00 @:0 B@:0
2021-05-08 12:43:50,609 - Recv:  T:29.50 /0.00 B:65.95 /0.00 @:0 B@:0
2021-05-08 12:43:52,609 - Recv:  T:29.63 /0.00 B:65.86 /0.00 @:0 B@:0
2021-05-08 12:43:54,609 - Recv:  T:29.63 /0.00 B:65.78 /0.00 @:0 B@:0
2021-05-08 12:43:56,609 - Recv:  T:29.63 /0.00 B:65.65 /0.00 @:0 B@:0
2021-05-08 12:43:58,609 - Recv:  T:29.63 /0.00 B:65.56 /0.00 @:0 B@:0
2021-05-08 12:44:00,609 - Recv:  T:29.75 /0.00 B:65.39 /0.00 @:0 B@:0
2021-05-08 12:44:02,610 - Recv:  T:29.75 /0.00 B:65.30 /0.00 @:0 B@:0
2021-05-08 12:44:04,610 - Recv:  T:29.88 /0.00 B:65.13 /0.00 @:0 B@:0
2021-05-08 12:44:06,610 - Recv:  T:29.88 /0.00 B:65.00 /0.00 @:0 B@:0
2021-05-08 12:44:06,612 - Recv: Error:Heating failed, system stopped! Heater_ID: bed
2021-05-08 12:44:08,617 - Changing monitoring state from "Operational" to "Error"
```
The Thermal & Watch settings I last used are
```
#define THERMAL_PROTECTION_BED_PERIOD        40 // Seconds - 20
#define THERMAL_PROTECTION_BED_HYSTERESIS     2 // Degrees Celsius - 2

#define WATCH_BED_TEMP_PERIOD                70  // Seconds  - 60
#define WATCH_BED_TEMP_INCREASE               2  // Degrees Celsius - 2
```

I have attached my config files if it helps....
[Config Files.zip](https://github.com/MarlinFirmware/Marlin/files/6445210/Config.Files.zip)

On a side note I have found it confusing and difficult to make sense of the wording describing the conditions that the two thermal bed settings use.