MarlinFirmware/Marlin #21803: [BUG] Compile error for Anycubic Deltas

DukeLander, 2021-05-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

In both example config files for Anycubic Kossel and Linear Plus (stable and bugfix) is:

// Feedrate (mm/min) for the first approach when double-probing (MULTIPLE_PROBING == 2)
#define Z_PROBE_FEEDRATE_FAST **_homing_feedrate_mm_m.z_**

which produces compiling error:
Compiling .pio\build\LPC1768\src\src\feature\bedlevel\bedlevel.cpp.o
In file included from Marlin\src\inc/MarlinConfig.h:47,
                 from Marlin\src\MarlinCore.h:24,
                 from Marlin\src\MarlinCore.cpp:31:
Marlin\src\inc/../../Configuration.h:1213:31: error: 'homing_feedrate_mm_m' was not declared in this scope
 1213 | #define Z_PROBE_FEEDRATE_FAST homing_feedrate_mm_m.z
...
...

If I change "homing_feedrate_mm_m.z" to any value, i.e. (100*60) I got:

Compiling .pio\build\LPC1768\src\src\feature\bedlevel\abl\abl.cpp.o
In file included from Marlin\src\MarlinCore.cpp:48:
Marlin\src\module/motion.h:76:52: error: conversion from 'int' to non-scalar type 'const xyz_feedrate_t' {aka 'const XYZval<float>'} requested
   76 | constexpr xyz_feedrate_t homing_feedrate_mm_m = (20*60)
      |                                                 ~~~^~~~
...
...

### Bug Timeline

2.0.7.2 compiles without error

### Expected behavior

Compiling succeed without errors

### Actual behavior

Error

### Version of Marlin Firmware

Latest and bugfix with latest config

### Printer model

Anycubic Delta Linear Plus

### Electronics

BTT SKR 1.3

### Additional information & file uploads

Thanks for great work!

shitcreek, 2021-05-05:
Issue confirmed.
`homing_feedrate_mm_m.z` is wrong and improper usage.
`#define HOMING_FEEDRATE_MM_M` needs to be enabled.
Once `HOMING_FEEDRATE_MM_M` is enabled, you can change `Z_PROBE_FEEDRATE_FAST` to whatever eg `(30*60)` and it will not throw a compiler error.

DukeLander, 2021-05-05:
After enabling #define HOMING_FEEDRATE_MM_M and putting some value i.e. to (50*60) in Z_PROBE_FEEDRATE_FAST, compile error:

Compiling .pio\build\LPC1768\src\src\feature\bedlevel\abl\abl.cpp.o
In file included from Marlin\src\feature\babystep.cpp:29:
Marlin\src\feature\../module/motion.h:76:52: error: conversion from 'int' to non-scalar type 'const xyz_feedrate_t' {aka 'const XYZval<float>'} requested
   76 | constexpr xyz_feedrate_t homing_feedrate_mm_m = **(20*60)**
      |                                                 ~~~^~~~
Compiling .pio\build\LPC1768\src\src\feature\bedlevel\bedlevel.cpp.o
In file included from Marlin\src\MarlinCore.cpp:48:
Marlin\src\module/motion.h:76:52: error: conversion from 'int' to non-scalar type 'const xyz_feedrate_t' {aka 'const XYZval<float>'} requested
   76 | constexpr xyz_feedrate_t homing_feedrate_mm_m = **(20*60)**
      |                                                 ~~~^~~~

I dont know from where comes bolded value, I have not such in Config.h

ellensp, 2021-05-05:
You must have done something wrong. update works fine...  give me a moment creating pr for it so you can see exact changes

ellensp, 2021-05-05:
see https://github.com/MarlinFirmware/Configurations/pull/485/files

ellensp, 2021-05-05:
I see no issues with Anycubic Kossel, just Anycubic Kossel Linear Plus Configs

ellensp, 2021-05-05:
PR created in Configurations, closing this issue.

closed 2021-05-05 by ellensp