MarlinFirmware/Marlin #21807: [BUG] AnyCubic i3 mega fails to compile without Z_MULTI_ENDSTOPS

jrmoserbaltimore, 2021-05-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin fails sanity check when using `Z_MULTI_ENDSTOPS` with `USE_PROBE_FOR_Z_HOMING`.

When disabling `Z_MULTI_ENDSTOPS`, Marlin compile fails:

```
Marlin/src/module/endstops.cpp: In static member function 'static void Endstops::report_states()':
Marlin/src/module/endstops.cpp:445:15: error: 'Z2_MIN_ENDSTOP_INVERTING' was not declared in this scope
     ES_REPORT(Z2_MIN);
```

### Bug Timeline

Worked in 2.0.7.2; failed in 2.0.8

### Expected behavior

It compiled in 2.0.7.2; I expected the configuration (patched to the 2.0.8 AnyCubic i3 Mega base configuration) to compile in 2.0.8.

### Actual behavior

A bunch of things went on with attempts to read the source code and tweak settings, none of which got the code to compile.

### Steps to Reproduce

Try to build Marlin (using platformio here, via Marlin Flasher 0.51 in Octoprint)

### Version of Marlin Firmware

2.0.8

### Printer model

AnyCubic i3 Mega Pro

### Add-ons

BLTouch added

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

[Marlin-2.0.8-config.zip](https://github.com/MarlinFirmware/Marlin/files/6424834/Marlin-2.0.8-config.zip) (not working, 2.0.8)

[Marlin-2.0.7.2-config.zip](https://github.com/MarlinFirmware/Marlin/files/6424838/Marlin-2.0.7.2-config.zip) (working, 2.0.7.2)

ellensp, 2021-05-05:
Is due to your motherboard having Z2 MIN pin defined

Changing Marlin/src/module/endstops.cpp
```
  #if HAS_Z2_MIN
    ES_REPORT(Z2_MIN);
  #endif

```
to
  
```
 #if HAS_Z2_MIN && ENABLED(Z_MULTI_ENDSTOPS)
    ES_REPORT(Z2_MIN);
  #endif
```

gets around the issue

ellensp, 2021-05-05:
actually this keeps interupts and other things working for  Z2_MIN

Changing Marlin/src/inc/Conditionals_post.h
```
#if PIN_EXISTS(Z2_MIN)
  #define HAS_Z2_MIN 1
#endif

```to
```
#if PIN_EXISTS(Z2_MIN) && ENABLED(Z_MULTI_ENDSTOPS)
  #define HAS_Z2_MIN 1
#endif
```
Is a lot cleaner

ellensp, 2021-05-05:
I didnt like either of those...

ended up changing Marlin/src/pins/ramps/pins_TRIGORILLA_14.h 
```    #define Z2_MIN_PIN                          43```
to
```  #if ENABLED(Z_MULTI_ENDSTOPS)
    #define Z2_MIN_PIN                          43
  #endif
```
This fixes the problem at the source of the issue.

ellensp, 2021-05-06:
A fix has been merged. Closing

closed 2021-05-06 by ellensp