MarlinFirmware/Marlin #21814: [BUG] error of building for Lerdge K with USB

sanek88lbl, 2021-05-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Don't build for LERDGEK_usb_flash_drive.

After last update there are new errors other than [it](https://github.com/MarlinFirmware/Marlin/issues/21302#issue-826874262).

### Actual behavior

```
...
Compiling .pio\build\LERDGEK_usb_flash_drive\src\src\HAL\STM32\timers.cpp.o
In file included from Marlin\src\HAL\STM32\tft\tft_fsmc.cpp:28:
Marlin\src\HAL\STM32\tft\tft_fsmc.h:55:12: error: 'SRAM_HandleTypeDef' does not name a type; did you mean 'UART_HandleTypeDef'?
   55 |     static SRAM_HandleTypeDef SRAMx;
      |            ^~~~~~~~~~~~~~~~~~
      |            UART_HandleTypeDef
In file included from Marlin\src\HAL\STM32\tft\tft_fsmc.cpp:28:
Marlin\src\HAL\STM32\tft\tft_fsmc.h:100:12: error: 'FSMC_NORSRAM_DEVICE' was not declared in this scope; did you mean 'FSMC_NORSRAM_TYPEDEF'?
  100 |   {PD_14,  FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D00
      |            ^~~~~~~~~~~~~~~~~~~
      |            FSMC_NORSRAM_TYPEDEF
Compiling .pio\build\LERDGEK_usb_flash_drive\src\src\HAL\STM32\usb_host.cpp.o
Marlin\src\HAL\STM32\tft\tft_fsmc.h:101:12: error: 'FSMC_NORSRAM_DEVICE' was not declared in this scope; did you mean 'FSMC_NORSRAM_TYPEDEF'?
  101 |   {PD_15,  FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D01
      |            ^~~~~~~~~~~~~~~~~~~
      |            FSMC_NORSRAM_TYPEDEF
Marlin\src\HAL\STM32\tft\tft_fsmc.h:102:12: error: 'FSMC_NORSRAM_DEVICE' was not declared in this scope; did you mean 'FSMC_NORSRAM_TYPEDEF'?
  102 |   {PD_0,   FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D02
      |            ^~~~~~~~~~~~~~~~~~~
      |            FSMC_NORSRAM_TYPEDEF
Compiling .pio\build\LERDGEK_usb_flash_drive\src\src\HAL\STM32\usb_serial.cpp.o
Marlin\src\HAL\STM32\tft\tft_fsmc.h:103:12: error: 'FSMC_NORSRAM_DEVICE' was not declared in this scope; did you mean 'FSMC_NORSRAM_TYPEDEF'?
  103 |   {PD_1,   FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D03
      |            ^~~~~~~~~~~~~~~~~~~
      |            FSMC_NORSRAM_TYPEDEF
compilation terminated due to -fmax-errors=5.
*** [.pio\build\LERDGEK_usb_flash_drive\src\src\HAL\STM32\tft\tft_fsmc.cpp.o] Error 1
============================================= [FAILED] Took 21.96 seconds =============================================

Environment              Status    Duration
-----------------------  --------  ------------
LERDGEK_usb_flash_drive  FAILED    00:00:21.965
======================================== 1 failed, 0 succeeded in 00:00:21.965 ======================================== 
```

### Steps to Reproduce

1. Clone last bugfix-2.0.x branch.
2. Replace _Configuration.h_ from my [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6427561/Configuration.zip)
3. Start Build from **_PROJECT TASKS -> LERDGEK_usb_flash_drive->General_**.

### Version of Marlin Firmware

Marlin 2.0.8 (May 5 2021)

### Printer model

Custom, B&R

### Electronics

Lerdge K board

rhapsodyv, 2021-05-08:
The same config builds fine on normal lerdge env?

rhapsodyv, 2021-05-08:
can you try this: https://github.com/rhapsodyv/Marlin/tree/fix-LERDGEX_usb_flash_drive ?

sanek88lbl, 2021-05-08:
> can you try this: https://github.com/rhapsodyv/Marlin/tree/fix-LERDGEX_usb_flash_drive ?

Thanks, but it didn't help(
I tried it:
```
....
Marlin\src\HAL\STM32\eeprom_wired.cpp: In static member function 'static size_t PersistentStore::capacity()':
Marlin\src\HAL\STM32\eeprom_wired.cpp:38:37: error: 'E2END' was not declared in this scope
   38 |   #define MARLIN_EEPROM_SIZE size_t(E2END + 1)
      |                                     ^~~~~
Marlin\src\HAL\STM32\eeprom_wired.cpp:40:45: note: in expansion of macro 'MARLIN_EEPROM_SIZE'
   40 | size_t PersistentStore::capacity() { return MARLIN_EEPROM_SIZE; }
      |                                             ^~~~~~~~~~~~~~~~~~
*** [.pio\build\LERDGEK_usb_flash_drive\src\src\HAL\STM32\eeprom_wired.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32\tft\tft_fsmc.cpp:28:
Marlin\src\HAL\STM32\tft\tft_fsmc.h:55:12: error: 'SRAM_HandleTypeDef' does not name a type; did you mean 'UART_HandleTypeDef'?
   55 |     static SRAM_HandleTypeDef SRAMx;
      |            ^~~~~~~~~~~~~~~~~~
      |            UART_HandleTypeDef
In file included from Marlin\src\HAL\STM32\tft\tft_fsmc.cpp:28:
Marlin\src\HAL\STM32\tft\tft_fsmc.h:100:12: error: 'FSMC_NORSRAM_DEVICE' was not declared in this scope; did you mean 'FSMC_NORSRAM_TYPEDEF'?
  100 |   {PD_14,  FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D00
      |            ^~~~~~~~~~~~~~~~~~~
      |            FSMC_NORSRAM_TYPEDEF
Marlin\src\HAL\STM32\tft\tft_fsmc.h:101:12: error: 'FSMC_NORSRAM_DEVICE' was not declared in this scope; did you mean 'FSMC_NORSRAM_TYPEDEF'?
  101 |   {PD_15,  FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D01
      |            ^~~~~~~~~~~~~~~~~~~
      |            FSMC_NORSRAM_TYPEDEF
Marlin\src\HAL\STM32\tft\tft_fsmc.h:102:12: error: 'FSMC_NORSRAM_DEVICE' was not declared in this scope; did you mean 'FSMC_NORSRAM_TYPEDEF'?
  102 |   {PD_0,   FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D02
      |            ^~~~~~~~~~~~~~~~~~~
      |            FSMC_NORSRAM_TYPEDEF
Marlin\src\HAL\STM32\tft\tft_fsmc.h:103:12: error: 'FSMC_NORSRAM_DEVICE' was not declared in this scope; did you mean 'FSMC_NORSRAM_TYPEDEF'?
  103 |   {PD_1,   FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D03
      |            ^~~~~~~~~~~~~~~~~~~
      |            FSMC_NORSRAM_TYPEDEF
compilation terminated due to -fmax-errors=5.
*** [.pio\build\LERDGEK_usb_flash_drive\src\src\HAL\STM32\tft\tft_fsmc.cpp.o] Error 1
================================================== [FAILED] Took 20.50 seconds ==================================================
Environment              Status    Duration
-----------------------  --------  ------------
LERDGEK_usb_flash_drive  FAILED    00:00:20.504
============================================= 1 failed, 0 succeeded in 00:00:20.504 =============================================
```

sanek88lbl, 2021-05-08:
> The same config builds fine on normal lerdge env?

Yes, it's builds fine on normal lerdge env. 
I think, USB pins are conflict with FSMC pins.

rhapsodyv, 2021-05-08:
This should works now: #21847

PR #21847 (2021-05-09): Fix Lerdge usb flash drive envs
Fix #21814 (missing lerdge flags for usb envs)