MarlinFirmware/Marlin #21815: [BUG] LERDGE_ST7796 driver error

sanek88lbl, 2021-05-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After last big update of _Marlin-bugfix-2.0.x_ `LERDGE_ST7796` driver don't work. Just `ST7796 `works, but changes colors.
It's build without errors and board connect to Octoprint host. Command `M111 S247` and `M115` do not give results.

### Actual behavior

Building in VSCode\Platformio:
```
> Executing task: C:\Users\user\.platformio\penv\Scripts\platformio.exe run --environment LERDGEK <

Processing LERDGEK (platform: ststm32@~12.1; board: marlin_STM32F407ZGT6; framework: arduino)
------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_STM32F407ZGT6.html
PLATFORM: ST STM32 (12.1.1) > STM32F407ZGT6(192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200819 (1.9.0)
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TMCStepper> 0.7.1
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
|-- <EEPROM> 2.0.1
Building in release mode
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\HAL.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\HAL_MinSerial.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\HAL_SPI.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\MarlinSPI.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\MarlinSerial.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\Sd2Card_sdio_stm32duino.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\Servo.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\eeprom_flash.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\eeprom_sdcard.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\eeprom_sram.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\eeprom_wired.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\fast_pwm.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\fastio.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\msc_sd.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\tft\tft_fsmc.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\tft\tft_ltdc.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\tft\tft_spi.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\tft\xpt2046.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\timers.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\usb_host.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\usb_serial.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\STM32\watchdog.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\shared\HAL_MinSerial.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\shared\eeprom_if_i2c.cpp.o
Compiling .pio\build\LERDGEK\src\src\HAL\shared\esp_wifi.cpp.o
[... 221 lines cut ...]
Recv: ok
```

### Steps to Reproduce

1. Clone bugfix-2.0.x branch.
2. Replace _Configuration.h_ from my [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6427921/Configuration.zip).
3. Build from **_PROJECT TASKS -> LERDGEK->General_**.
4. Flash your Lerdge K/X board.

### Version of Marlin Firmware

Marlin 2.0.8 (May 5 2021)

### Printer model

Custom, B&R

### Electronics

Lerdge K board

### Host Software

OctoPrint

rhapsodyv, 2021-05-08:
Try adding this:
`#define TFT_COLOR TFT_COLOR_RGB`

sanek88lbl, 2021-05-08:
> Try adding this:
> `#define TFT_COLOR TFT_COLOR_RGB`

Thanks, but it didn't help(
But I fixed the problem: [fix display](https://github.com/MarlinFirmware/Marlin/pull/21812/commits/ae6589bac1761e4abbbebc54efe2090382a94cf0).

closed 2021-05-08 by thinkyhead