MarlinFirmware/Marlin #21817: [BUG] (bug summary)

roastypup, 2021-05-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Used both stable and bug fix version of marlin while trying to edit creality board 4.2.7 silent board for ender 5.
trying to upgrade from my loud factory board to this silent board. creality firmware works, but the extrudent is on opposite polarities.

and this is the final bug that is stumping me, when trying to compile it. anyone know any fixes for this?

using this video so far to help me edit the code from teaching tech
https://www.youtube.com/watch?v=neS7lB7fCww&t=627s&ab_channel=TeachingTech

In file included from c:\users\[USER]\appdata\local\temp\arduino_build_246078\sketch\src\pins\stm32f1/pins_CREALITY_V4.h:27:0,
                 from c:\users\[USER]\appdata\local\temp\arduino_build_246078\sketch\src\pins\stm32f1/pins_CREALITY_V427.h:44,
                 from c:\users\[USER]\appdata\local\temp\arduino_build_246078\sketch\src\pins\pins.h:548,
                 from c:\users\[USER]\appdata\local\temp\arduino_build_246078\sketch\src\inc\marlinconfig.h:34,
                 from sketch\src\libs\BL24CXX.cpp:20:
c:\users\[USER]\appdata\local\temp\arduino_build_246078\sketch\src\pins\stm32f1/env_validate.h:25:4: error: #error "Oops! Select an STM32F1 board in 'Tools > Board.'"
   #error "Oops! Select an STM32F1 board in 'Tools > Board.'"
    ^~~~~
sketch\src\libs\BL24CXX.cpp:30:10: fatal error: libmaple/gpio.h: No such file or directory
 #include <libmaple/gpio.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Sanguino.

### Version of Marlin Firmware

marlin 2.0.x  config 2.0.8

### Printer model

ender 5

### Electronics

creality silent board 4.2.7 

### Add-ons

N/A

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

thisiskeithb, 2021-05-05:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-05-05 by thisiskeithb

ethanhjennings, 2021-05-13:
Is this actually a bug? I'm having the same issue when compiling for Ender 3 Pro V1.1.5 board from the Arduino IDE.

ellensp, 2021-05-13:
The original post is user error, They upgraded their motherboard to a stm32 based controller but continued to try and compile for a  ATmega1284P. 

Seek help from one of the resources listed above.

ethanhjennings, 2021-05-14:
Apologies if this is too off topic now, but I solved my issue. I was dumb and using the wrong configuration headers. The "CrealityV1" headers worked after changing the stepper drivers to TMC2208_STANDALONE. I also still had weird compiler errors that went away after switching to Platform IO.