MarlinFirmware/Marlin #21824: [FR] M106 / M107 with dual nozzle and 1 fan on a board with 2 fan connectors

cprobst87, 2021-05-06:
### Is your feature request related to a problem? Please describe.

Hello,
in M106_M107.cpp there ist the following code:
`#if ENABLED(SINGLENOZZLE)
  #define _ALT_P active_extruder
  #define _CNT_P EXTRUDERS
#else
  #define _ALT_P _MIN(active_extruder, FAN_COUNT - 1)
  #define _CNT_P FAN_COUNT
#endif`

So if you switch to T1, fan number 2 will be controlled by M106/M107, if you switch back to T0, fan number 1 will be controlled.

That's fine for users with 2 fans... I have only 1 fan, but I've 2 connectors on my board. So I had to change the code to:
`#else
  //#define _ALT_P _MIN(active_extruder, FAN_COUNT - 1)
  #define _ALT_P 0
  #define _CNT_P FAN_COUNT
#endif`

### Describe the feature you want

An option in configuration file for M106_M107 for the variable "_ALT_P", or "FAN_COUNT"... like "FAN_USE"

thisiskeithb, 2021-05-06:
> That's fine for users with 2 fans... I have only 1 fan, but I've 2 connectors on my board.

It sounds like you didn’t set `NUM_M106_FANS` correctly in your config.

cprobst87, 2021-05-06:
Thank you! I havn't seen this...

closed 2021-05-06 by cprobst87