MarlinFirmware/Marlin #21825: [FR] BLTouch auto-clearance

jrmoserbaltimore, 2021-05-06:
### Is your feature request related to a problem? Please describe.

Bed leveling is faster with `Z_CLEARANCE_MULTI_PROBE = 2`, but if the printer thinks the bed is a bit too low when it probes, the pin will not come out far enough to pass the trigger point, and it'll throw an error.  It works with  `Z_CLEARANCE_MULTI_PROBE = 3` but is slower.

### Are you looking for hardware support?

BLTouch

### Describe the feature you want

Upon a probing error, back off clearance, bit, reset the probe, and try again.  If this goes too far, just give up.  Raise 1.5mm higher between probes to ensure the probe doesn't drag against the board.

Possibly estimate how far away the board should be, deploy the probe, and see if you get a hit or error immediately; if so, back off a bit, reset the probe (if error), and try again.  This allows multiple probes just barely moving across the trigger point, taking several samples quickly rather than with 1+ second between them.  With 25 probe points and 2 additional probes (the first is always slow, extra clearance needed) that can save a minute or so on probing.  It's a bit more extreme when using UBL's 100-point probing pattern.

jrmoserbaltimore, 2021-05-06:
Okay, with the Z axis sufficiently unlevel, `G34` outright fails at 2mm clearance because it hits the bed and can't get past the trigger point; and at 3mm clearance fails because the probe never reaches the bed and it gives up.

Had to manually adjust the Z axis to make it even work.

ellensp, 2021-05-07:
perhaps a pre probe run would be good.  Test 4 corners and centre and get a overall slope of the bed

jrmoserbaltimore, 2021-05-07:
Part of the problem was my Z gantry was sloped and it needed to level that.  At a probe clearance of 2mm, the left side hit the bed and then didn't retract back up above the trigger, so it failed; at 3mm, that worked, but then on the right side the probe couldn't find the bed.  I had to manually turn the right lead screw and then run `G34`.

I imagine it would fail the same way if it tried to touch all 4 corners.