MarlinFirmware/Marlin #21830: BLTouch doesn't deploy with LCD Home

coelholm, 2021-05-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

This week I update to bugfix-2.0.x to last commit (old commit was from Dec 9/2020). After that I had to struggle a little to get BLTouch working again, thanks for all comments, very helpful. Now, I have Auto Home working with G28 and BLTouch deploys normally.
I can't make Auto Home from LCD menu to work. It always home XY correctly and fails Z homing because BLTouch never deploys.

### Bug Timeline

Looks like an old issue

### Expected behavior

LCD Menu
Motion
Auto Home
 - Home X
 - Home Y
 - Move to center of Bed (Safe z home)
 - Deploy bltouch probe
 - Home Z

### Actual behavior

LCD Menu
Motion
Auto Home
 - Home X
 - Home Y
 - Move to center of Bed (Safe z home)
 - Home Z (Crash on bed)

### Steps to Reproduce

LCD Menu
Motion
Auto Home

### Version of Marlin Firmware

Marlin bugfix-2.0.x 

### Printer model

Ender 3

### Electronics

SKR Mini E3 V1.2

### Add-ons

Bigtreetech TFT35 E3 V3 Firmware(BIGTREE_TFT35_V3.0_E3.27.bin)

### Host Software

OctoPrint

thinkyhead, 2021-05-07:
You didn't include your configurations, so we're a little stuck.

coelholm, 2021-05-07:
Configuration.h and Configuration_adv.h
[lmc-cfg.zip](https://github.com/MarlinFirmware/Marlin/files/6444426/lmc-cfg.zip)

coelholm, 2021-05-07:
> You didn't include your configurations, so we're a little stuck.

closed 2021-07-19 by github-actions[bot]