MarlinFirmware/Marlin #21831: [FR] Are there options for paid programming/features/support

Brampie66, 2021-05-07:
### Is your feature request related to a problem? Please describe.

Hi,

I don't have a specific feature request right now actually.
But I would like to know what the options are toward some sort of paid support or programming.

I'm aware I can donate, and this may be a very good idea, but suppose I want a specific thing added to Marlin or I need help.
Are there ways to get someone to fix one specific issue or add a feature and just pay for the time spent.
This way users can be helped more directly while also contributing to Marlin.

Is anyone offering this ?

thx

### Are you looking for hardware support?

none

### Describe the feature you want

more options I guess

zonalimitatore, 2021-05-07:
Yes.Look to fiverr

thisiskeithb, 2021-05-07:
I’ve seen [IssueHunt](https://issuehunt.io/) used before and it looks like it’d be easy to tie into the existing bug/feature list.

Brampie66, 2021-05-11:
Thx for your input
I think IssueHunt would be a nice addition, it would also make it more clear which features are more important to users.
It is something the Marlin team needs to setup ?

Fiverr also looks like a good way to get help/support, difficult to say which offer would be able to provide additional features.
The focus seems to be getting a machine to run