MarlinFirmware/Marlin #21834: [BUG] Strange behavior of X-Axis (dual-endstop) while Auto-Bed-Leveling

reindeer-engineer, 2021-05-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi Guys, I attached a MK8 extruder/hotend to my [MPCNC Primo](https://docs.v1engineering.com/mpcnc/intro/). I added a inductive sensor to the tip of the z-axis as a probe. it has a dual-x-carriage and dual-y-carriage. auto squaring works quite good, but i ran into a problem with auto bed leveling. when i send the G29 command to start the automated bed leveling, the x-axis moves proper into the positiv direction (on negative range), but moves way to less or even not into the negative direction.

here you can see the behavior: [https://www.youtube.com/watch?v=E6EE5i1ZD-Q](https://www.youtube.com/watch?v=E6EE5i1ZD-Q)

has any one an idea, why it is acting like that? 
does anybody else use the dual-endstop setup with ABL successfull? 
if yes, would you please share you config with me??

I am using the [dual endstop setup ](https://docs.v1engineering.com/electronics/dual-endstops/) with a 6th/external stepper driver for the extruder on a ramps1.4 board + an arduino mega + 6x drv8825.
ezgif.com-gif-maker

my modifications in Configuration.h  :
```
#define TEMP_SENSOR_0 1
#define X_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // Set to true to invert the logic of the probe.
#define FIX_MOUNTED_PROBE
#define NOZZLE_TO_PROBE_OFFSET { 37, 20, 0 }
#define XY_PROBE_SPEED 1000
//#define MIN_SOFTWARE_ENDSTOPS
#define AUTO_BED_LEVELING_LINEAR
#define RESTORE_LEVELING_AFTER_G28
#define GRID_MAX_POINTS_X 5
```

pins_RAMPS.h
```
#define E0_STEP_PIN 42
#define E0_DIR_PIN 40
#define E0_ENABLE_PIN 63
```

### Bug Timeline

a new bug

### Expected behavior

when executing a G29 Command I excepted the probe to measure build plate in a 3x3 mesh

### Actual behavior

here you can see the behavior: [https://www.youtube.com/watch?v=E6EE5i1ZD-Q](https://www.youtube.com/watch?v=E6EE5i1ZD-Q)

when the probe is bed leveling, it is moving like shown in the following table:

Position | X on LCD | real X | Y
1           34                34                34
2           68                68                34
3           102              102              34
4           102              102              68
5           68                100+-4         68
6           34                98+-4           68
7           34                98+-4           68
8           68                131+-4           102             // here it moves out of the defined build plate
9           102              164+-4           102

### Steps to Reproduce

plfüwlüdf

### Version of Marlin Firmware

2.0.8

### Printer model

MPCNC

### Electronics

Arduino + Ramps1.4 + DRV8825

### Add-ons

RepRapLCD, Inductive Probe

### Your Slicer

Cura

### Host Software

OctoPrint

thisiskeithb, 2021-05-14:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

tombrazier, 2022-02-24:
@thisiskeithb I think #21965 looks like a duplicate of this and it has a lot more information. This issue has not had an update since May 2021. Should it be closed?

thisiskeithb, 2022-02-24:
> Should it be closed?

OP never responded, so that works for me.

closed 2022-02-24 by thisiskeithb

thinkyhead, 2022-04-21:
Duplicate of #21965