MarlinFirmware/Marlin #21841: [FR] (feature summary)

osckylp, 2021-05-08:
### Is your feature request related to a problem? Please describe.

I need a whistle after warm-up, but the marlin doesn't have that option.

### Are you looking for hardware support?

ramps 1.4

### Describe the feature you want

It would be nice if there is an audible signal after preheating, so I can preheat and be close but not have to look at the screen at all times.

ellensp, 2021-05-08:
What are you referring to exactly? There any many things that can be considered preheat...

Are you selecting preheat from the advanced menu on a LCD?

osckylp, 2021-05-08:
> ¿A qué te refieres exactamente? Hay muchas cosas que pueden considerarse precalentamiento ...
> 
> ¿Está seleccionando precalentar en el menú avanzado de una pantalla LCD?
when preheating the bed or hotend from the lcd temperature menu (Sorry for my English, I speak Spanish and I use the google translator)

borland1, 2021-05-09:
This could be done with your slicer's Start Code,
```
M190 S80 ; Wait for 80C Bed Temperature
M300 S440 P200 ; Play Tone
M300 S660 P250
M300 S880 P300
```

YotaPower04, 2023-02-15:
to further expand on this problem i use the built in preheat faction on my printer on various occasions to do things like change filament, change nozzles, and leveling the print bed. I'm running my printers in a print farm type scenario so my attention is easily swayed from what I'm currently waiting on to what I could be doing while I'm waiting. (I'm doing it currently looking for a solution to this problem.... brb I need to change filament) it would be nice if my machine could give me and audible beep when the default preheat function built into firmware has reached the desired temp