MarlinFirmware/Marlin #21844: [FR] Manual mesh bed levelling, ability to go through already calibrated points

JeanLucCoulon, 2021-05-08:
### Is your feature request related to a problem? Please describe.

This applies to a Biqu B1.

When I do a manual mesh bed levelling, it goes through the different points.
If I missed a point, I have to restart from the beginning.
This occurs quite often: if I press the "next" button twice, which is [too] easy with the touch screen, I miss a point, and I've to restart from the beginning.

### Are you looking for hardware support?

Biqu B, TFT 35

### Describe the feature you want

I would like to be able to go through the different already calibrated points.
Or the system to ignore a double press of the button.

borland1, 2021-05-09:
There are settings for the touch screen sensitivity in `Configuration.h` file.  Are these not working for you? 
```
#if ENABLED(TOUCH_SCREEN)
  #define BUTTON_DELAY_EDIT  50 // (ms) Button repeat delay for edit screens
  #define BUTTON_DELAY_MENU 250 // (ms) Button repeat delay for menus
```

borland1, 2021-05-10:
That is for use with `// ADS7843/XPT2046 ADC Touchscreen such as ILI9341 2.8`

Sorry if I mislead you on that, but you did not provide your config files.  If your TFT35 is a serial (AUX-1) touch screen, then you may need to update the display's firmware.   [https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware](url)

JeanLucCoulon, 2021-05-11:
I've a Biqu B1. I've updated the touch screen firmware for the latest version (namely BIGTREE_TFT35_V3.0_B1.27.CUR).

qystan, 2021-08-04:
Will be useful if the requested feature can be done in Edit Mesh so that instead of editing numbers which has no reference, move the nozzle to the probe point and adjustment is done by moving the nozzle and using a paper to test.

sarvenn, 2021-08-28:
A "back" option will be good nearby "next".